cyberbotics/webots #3534: Rendering difference between WebotsJS/X3D and Webots

omichel, 2021-08-03:
The FESTO logo visible on the robot web component at https://cyberbotics.com/doc/guide/robotino3 looks bad where it looks good when running the robotino3.wbt simulation in Webots.

On the web (Firefox/Window):
[image]

In Webots:
[image]

Where does this difference come from? Could it be the resolution of the depth buffer for depth test?

BenjaminDeleze, 2021-11-22:
I investigated the following possibilities:

- Resolution of the depth buffer: both version are using the same shaders and I could not find a difference in the way the depth buffer are created.
- Problem in indexedFaceSet/Mesh: It is not the case because I was able to reproduce the issue with only a box and a cylinder

It seems like the issue appears when the position difference between two textures is less than 0.03mm.

omichel, 2021-11-22:
> Resolution of the depth buffer: both version are using the same shaders and I could not find a difference in the way the depth buffer are created.

Is the depth buffer 32-bit or 24-bit on both implementations? Is it combined with the stencil buffer (8-bit) used for shadows?

omichel, 2021-11-22:
See for example https://stackoverflow.com/questions/22398575/depth-stencil-framebuffer-concerns

BenjaminDeleze, 2021-11-22:
> Is the depth buffer 32-bit or 24-bit on both implementations? Is it combined with the stencil buffer (8-bit) used for shadows?

Yes both use the `WR_TEXTURE_INTERNAL_FORMAT_DEPTH24_STENCIL8`

omichel, 2021-11-22:
> Yes both use the `WR_TEXTURE_INTERNAL_FORMAT_DEPTH24_STENCIL8`

And is the depth test the same? Also the value of the near and the far for the depth buffer are the same?
This article may be inspiring: https://learnopengl.com/Advanced-OpenGL/Depth-testing

BenjaminDeleze, 2021-11-23:
I found out that the view and projection matrix are not totally the same in Webots and WebotsJS. Seems like a precision difference of glm-js or of the x3d export. However, I could not prove yet that this is the cause of the problem. I will continue to investigate it.

BenjaminDeleze, 2021-11-24:
The precision difference comes from the x3d export. In `src/webots/vrml/wbSFRotation.hpp` and `src/webots/vrml/wbSFVector3.hpp` the precision of the x3d export is `FLOAT_MAX`. But it does not impact the rendering so we should keep it like it is to keep the export light.

BenjaminDeleze, 2021-11-24:
Half of the problem is solved:

The clipping difference between cube and cylinder is due to `glClipControl` (see #3101). If I disable glClipControl on Webots I get the same result.

However, it does not change anything for the robotino problem or for clipping edge cases between two IFS

BenjaminDeleze, 2022-06-10:
Close because similar of #3101

closed 2022-06-10 by BenjaminDeleze