cyberbotics/webots #3547: Webots <-> Controller protocol

ShuffleWire, 2021-08-05:
Disclamer : I'm asking to improve my own coding style, and Webots show me new way of doing things, and I would like to understand why thing are done in such way.

After digging in the code about Webots / Controller communication, I understood that it work this way : add each variable one after the other on a stream, send it the stream, and decode it using the sames types and order, using getters.

When working on personal project, I'm used to doing this using structures made for that, shared at build time between both side : I fill a structure member per member, dump it into a byte array (`raw_data = (uint8_t*)&my_struct`), send the resulting array, and converting it back to the same method (`my_struct = *(my_struct_t*)raw_data`).
I'm using that for three reasons : no error in term of size nor data organization, readability and performance (no individual function call).
The main drawback is in term of byte-order, or eventually data size (even if standard data type (stdint for instance) could help on that). However, because I'm programming software built and ran on the same PC or on "compatible" hardware, is not an issue (it's like Webots and controller)
Moreover I'm usually working in pure C, so don't know about C++, Matlab implementation. However I've been able to make it (kinda) work in Python. but not sure how work others API, do they relies on the C one to pass data ? After a quick look, it look like it, but not sure.

I'm wonder what was the rational behind doing the way Webots do ?

lukicdarkoo, 2021-08-06:
> do they relies on the C one to pass data

All APIs (C++/Python/Java/MATLAB/ROS) rely on C. The higher-level APIs simply call C functions.

> The main drawback is in term of byte-order, or eventually data size

libController and Webots are compiled with different compilers. libController with gcc and Webots with g++ (clang on macOS). See [this](https://stackoverflow.com/questions/48068142/gcc-and-g-struct-packing-differences) for example.

@stefaniapedrazzi and @omichel can probably give you more comprehensive answers.

omichel, 2021-08-11:
Actually, not all the parameters of an object need to be sent from Webots to libController, this is why we do not serialize the whole object, but rather some of the fields of that object that we want to pass to the libController. I hope this helps you to better understand the way it is implemented. In any case, there should not be any byte-order problem between C++ and C on the same machine, as byte order is just CPU-dependent and not language-dependent.

closed 2021-08-11 by omichel