cyberbotics/webots #3554: C controller arguments format

stefaniapedrazzi, 2021-08-09:
From the [documentationt](https://www.cyberbotics.com/doc/reference/robot) it seems that it is sufficient to add a space between words to separate the arguments.
However, this doesn't work in C controllers where each MFString item of the `controllerArgs` is parse as a single argument.
Only if each argument is given as single MFString item the result is the expected one.

For example:
* ``controllerArgs ["opt1 opt2"]`` will result in:
   ``argc = 2; argv = [<controller_name>, "op1 opt2"]``

* ``controllerArgs ["opt1", "opt2"]`` will result in:
   ``argc = 3; argv = [<controller_name>, "op1", "opt2"]``

It would be more intuitive if the behavior will be the documented one, where it is sufficient to add a space to separate arguments.
Writing the arguments in a single or multiple MFString items should not affect the behavior, that should be exactly the same.

omichel, 2021-08-11:
I believe that having spaces in arguments is useful, like if you want to pass the path to a file that contains spaces. If we consider spaces in arguments as separators, there is no way to specify spaces in an argument any more (or we should use escape codes, like backslash, which is not ideal). So, I would simply fix the documentation.

stefaniapedrazzi, 2021-08-11:
The best option really depends on the use case: many arguments (requiring adding a new item per each argument) vs spaces in the argument (requiring escape codes).

What I don't like too much and that is not intuitive is that the format in the controller differs from what you would write when starting the same program in a terminal (where arguments are simply divided by spaces and spaces needs to be escaped).

But it is fine to just fix the documentation and properly document it.

closed 2021-08-11 by stefaniapedrazzi