cyberbotics/webots #3556: Sumo still crashing in `highway_overtake.wbt`

ad-daniel, 2021-08-10:
**Describe the Bug**
There appears to still exist problems with sumo. 

Steps to reproduce:
- Open  `highway_overtake.wbt`
- Let it run a lot. It might take a simulated 30 minutes, sometimes 1 hour before it crashes.

It doesn't appear to be a regression, the crash fixed [here](https://github.com/cyberbotics/webots/pull/3391) happened almost immediately, so this one might have gone unnoticed.

The crash affects both ubuntu 20 and windows

llessieux, 2021-09-29:
Stack overflow for me and I didn't need to wait for 20 minutes.

llessieux, 2021-09-30:
caused by a NaN value being generated apparently and set to the angle section of a WbSFRotation object. No idea where yet.
This cause equality checks to fail all the time, triggering an infinite update loop causing the stack overflow.
I have found out that minimizing the windows make it happen faster too ;)

llessieux, 2021-09-30:
The cause is a floating point error it seems.
In WbAbstractTransform::rotate when the rotation angle is very small it seems that it can caused an invalid computation.
The WbRotation isValid is not checked (and it doesn't check for NaN on the angle anyway) so it propagates an invalid value and that causes havoc.

llessieux, 2021-09-30:
Funny I have driven on that road before :)

llessieux, 2021-09-30:
I'll propose a fix to avoid the crash but I can't really tell how wide an effect that can have.
I guess everybody can agree that propagating NaN values are not a good idea but what should be done in these case can be hard to decide.
The top call I can see is the WbSolid::prePhysicsStep. The rotate call is where it goes wrong.

omichel, 2021-09-30:
Maybe can you dig deeper to understand how this NaN value is computed? Is it the result of some specific mathematical operation?

llessieux, 2021-09-30:
Yes I should be able to get to the root of the issue. My guess is that the normalize of the angle is the issue but I will try to get the exact values causing the problem.

llessieux, 2021-09-30:
Invalid WbRotation created: 
v:0 6.0729265702774704e-05 0
Rotation :[
  0.9999999981559782 0 6.072926566544601e-05
  0 1 0
  -6.072926566544601e-05 0 0.9999999981559782
]

Intermediate Matrix WbMatrix3(mRotation->value()):[
  -0.9999999981559783 0 6.07292633151153e-05
  0 0.9999999999999994 0
  -6.07292633151153e-05 0 -0.9999999981559783
]

Intermediate Matrix rotation * WbMatrix3(mRotation->value()):[
  -1 0 -2.3503307115748003e-12
  0 0.9999999999999994 0
  2.3503307115748003e-12 0 -1
]
value passed to Acos : -1.0000000000000002  <<< This is the problem.
mRotation :0 0.9999999999999999 0 3.141531924326441
 Cos(mRotation.angle()) : -0.9999999981559783
 Sin(mRotation.angle()) : 6.0729263315115305e-05
newRotation: 0 1 0 nan
v,mRotation,newRotation values (binary): 
000000000000000000c04aaef1d60f3f00000000000000000000000000000000ffffffffffffef3f00000000000000007a93527ddb2109400000000000000000000000000000f03f0000000000000000000000000000f8ff

Acos definition:
NaN if d < -1 or d > 1 or d equals NaN.
So yes -1.0000000000000002  will cause an acos.
I propose to clamp the value passed to acos between -1 and 1.

llessieux, 2021-09-30:
And there are quite a few places in the code where acos is used and no check are done.
Potential other candidates. 
Funny enough, the WbRotation::fromQuaternion has the check to properly handle that already.

llessieux, 2021-09-30:
I can now confirm that clamping the value passed to acos in WbRotation::fromMatrix3 fixes the problem.
I would like to propose adding a clampedAcos function in WbMathsUtilities and replace all exiting calls to acos by the new function to prevent other occurences in the other locations using acos.

omichel, 2021-09-30:
Yes, that sounds good. Please go ahead with the implementation and propose a PR. I will be happy to review it.

llessieux, 2021-09-30:
How do you want me to classify the issue/branch ?
Does fix-acos-issues sounds good to you?

omichel, 2021-09-30:
> Does fix-acos-issues sounds good to you?

Yes, very good.

llessieux, 2021-10-01:
Hi Olivier, so it turned out that a few other places had already checks in place for that.

There are also two cases of asin not checked.
One in WbCylinder::pickUVCoordinate and one in WbMatrix3::toEulerAnglesZYX.
Should I protect these too?

I think that the atan2 cases are mostly fine as long as they don't get infinity in them. So I propose to leave these alone for now until someone find a case where they go wrong ;)

omichel, 2021-10-01:
> There are also two cases of asin not checked.
> One in WbCylinder::pickUVCoordinate and one in WbMatrix3::toEulerAnglesZYX.
> Should I protect these too?

Yes, that would be nice.

> I think that the atan2 cases are mostly fine as long as they don't get infinity in them. So I propose to leave these alone for now until someone find a case where they go wrong ;)

Sounds good.

closed 2021-10-01 by llessieux

ShuffleWire, 2021-10-03:
I'm sorry to read this commit that late, but I don't agree on how this is handle :/ Clamping give bad mathematical behavior (ie acos(cos(x)) != x [2pi]).
If the input value are very close to -1 or 1 (ie like 1.000001), that fine, but if suddenly the value is like huge (ie 15), it might hide a previous issue (not normalizing vector before a cross product, for instance).
Moreover, I do actually think that propagating NAN is a rather nice idea, as tracking it it usually an easy job, until we find the initial faulty mathematical operation.
Maybe add an assert in the clamping function, that way we could track down future issues.
Again, sorry to be so late, the overall idea behind this bugfix is nice :)

omichel, 2021-10-04:
@ShuffleWire: I basically agree with your comment and I believe adding an assert would be a nice idea. May I let you propose this in a PR?

llessieux, 2021-10-04:
I don't mind updating the code ;) 
So add an assert checking that the value is within the range with some epsilon value?
Does that sounds ok to everybody?

Propagating the NaN will just break everything and it will take a while to figure out where it went wrong because the side effect can be quite far from the cause ;) 
Took me half a day (but then again I am on Windows and using Visual Studio to debug so perhaps not the best way ;) to get to the root of the issue.

omichel, 2021-10-04:
> Does that sounds ok to everybody?

Sounds good to me.