cyberbotics/webots #3557: Remove warnings about missing shadows

ad-daniel, 2021-08-10:
**Describe the Bug**
After the change https://github.com/cyberbotics/webots/pull/3442, some worlds now display a warning:

- [ ] `highway_overtake.wbt`:
 `WARNING: StraightRoadSegment (PROTO) > CrashBarrier (PROTO) > Shape > IndexedFaceSet: Too many triangles (110000) in mesh: unable to cast shadows, please reduce the number of triangles below 21845 or set Shape.castShadows to FALSE to disable this warning.`

stefaniapedrazzi, 2021-08-16:
For info: we have a script [`tests/manual_tests/test_worlds_warning.py`](https://github.com/cyberbotics/webots/blob/master/tests/manual_tests/test_worlds_warnings.py) to check if any distributed world is displaying warnings at load.
I'm running it (takes quite long) to check if other warnings are detected.

stefaniapedrazzi, 2021-08-16:
I found a couple of other worlds raising issues related to shadows:
* `projects/vehicles/worlds/CH_Morges.wbt`
* `projects/vehicles/worlds/CH_Vens.wbt`
-> warnings printed when loading the `Forest` PROTO

PR #3578 (2021-08-16): Fixed forest shadow
Fixes #3557 (fully).

The trees should actually never cast shadows as they are semi-transparent.
closed 2021-08-16 by omichel