cyberbotics/webots #3572: Simulation freezing

felsager, 2021-08-13:
**Describe the bug**
When running a simulation of the Tiago Base robot the simulation freezes - the simulation speed goes to 0.00x. But no crash messages. Resetting the simulation gives the warning "WARNING: participant_controller: Forced termination (because process didn't terminate itself after 1 second)."

**Steps to reproduce**
This occurs with a basic time step of 16. 
Reducing the basic time step to 8 resolves the issue, and the simulation does not freeze.

**Screenshots**
Screenshot of gui when simulation is frozen:
[image]

**System**
 - Operating System: Linux Ubuntu 20.04.2

omichel, 2021-08-13:
It looks like the freeze takes place in one of your controller programs, not in Webots. If you replace your controllers with the "void" controller, do you still get the freeze?

felsager, 2021-08-13:
No using the "void" controller does not freeze the simulation. But if the problem has to do with the controller i wonder why it should be able to cause the simulation to freeze, and decreasing the basic time step size to 8 stops it from freezing

ad-daniel, 2021-08-13:
I can happen if you have an infinite loop. Can you share the world/controller? Or create a minimal example that showcases the issue?

lukicdarkoo, 2021-08-13:
Also, to confirm whether the controller is blocking the simulation you can turn off the synchronization (see the `synchronization` parameter [here](https://cyberbotics.com/doc/reference/robot))

felsager, 2021-08-13:
Hi again
Turning off synchronization made me able to debug the controller and see what actually went wrong with it, and was able to resolve an infinite loop happening in the planner. Thanks a lot everybody

closed 2021-08-13 by felsager