cyberbotics/webots #3574: Cannot download online resources in webots2021b

dzywater, 2021-08-15:
WARNING: DEF ELEVATION_GRID Solid > Shape > SandyGround (PROTO) > ImageTexture: Cannot download https://raw.githubusercontent.com/cyberbotics/webots/R2021b/projects/appearances/protos/textures/sandy_ground/sandy_ground_base_color.jpg: Error opening https://raw.githubusercontent.com/cyberbotics/webots/R2021b/projects/appearances/protos/textures/sandy_ground/sandy_ground_base_color.jpg
WARNING: DEF ELEVATION_GRID Solid > Shape > SandyGround (PROTO) > ImageTexture: Cannot download https://raw.githubusercontent.com/cyberbotics/webots/R2021b/projects/appearances/protos/textures/sandy_ground/sandy_ground_roughness.jpg: Error opening https://raw.githubusercontent.com/cyberbotics/webots/R2021b/projects/appearances/protos/textures/sandy_ground/sandy_ground_roughness.jpg
WARNING: DEF ELEVATION_GRID Solid > Shape > SandyGround (PROTO) > ImageTexture: Cannot download https://raw.githubusercontent.com/cyberbotics/webots/R2021b/projects/appearances/protos/textures/sandy_ground/sandy_ground_normal.jpg: Error opening https://raw.githubusercontent.com/cyberbotics/webots/R2021b/projects/appearances/protos/textures/sandy_ground/sandy_ground_normal.jpg
WARNING: DEF ELEVATION_GRID Solid > Shape > SandyGround (PROTO) > ImageTexture: Cannot download https://raw.githubusercontent.com/cyberbotics/webots/R2021b/projects/appearances/protos/textures/sandy_ground/sandy_ground_occlusion.jpg: Error opening https://raw.githubusercontent.com/cyberbotics/webots/R2021b/projects/appearances/protos/textures/sandy_ground/sandy_ground_occlusion.jpg
WARNING: Sassafras (PROTO) > Shape > PBRAppearance > ImageTexture: Cannot download https://raw.githubusercontent.com/cyberbotics/webots/R2021b/projects/objects/trees/protos/textures/sassafras_trunk.jpg: Error opening https://raw.githubusercontent.com/cyberbotics/webots/R2021b/projects/objects/trees/protos/textures/sassafras_trunk.jpg
WARNING: Sassafras (PROTO) > Shape > PBRAppearance > ImageTexture: Cannot download https://raw.githubusercontent.com/cyberbotics/webots/R2021b/projects/objects/trees/protos/textures/sassafras_leaves.png: Error opening https://raw.githubusercontent.com/cyberbotics/webots/R2021b/projects/objects/trees/protos/textures/sassafras_leaves.png
WARNING: Sassafras (PROTO) > Shape > PBRAppearance > ImageTexture: Cannot download https://raw.githubusercontent.com/cyberbotics/webots/R2021b/projects/objects/trees/protos/textures/sassafras_leaves_normal.jpg: Error opening https://raw.githubusercontent.com/cyberbotics/webots/R2021b/projects/objects/trees/protos/textures/sassafras_leaves_normal.jpg
WARNING: Moose (PROTO) > DEF MOOSE_BODY_SHAPE Transform > Shape > GlossyPaint (PROTO) > ImageTexture: Cannot download https://raw.githubusercontent.com/cyberbotics/webots/R2021b/projects/appearances/protos/textures/matte_car_paint/matte_car_paint_base_color.jpg: Error opening https://raw.githubusercontent.com/cyberbotics/webots/R2021b/projects/appearances/protos/textures/matte_car_paint/matte_car_paint_base_color.jpg
WARNING: Moose (PROTO) > DEF MOOSE_BODY_SHAPE Transform > Shape > GlossyPaint (PROTO) > ImageTexture: Cannot download https://raw.githubusercontent.com/cyberbotics/webots/R2021b/projects/appearances/protos/textures/glossy_car_paint/glossy_car_paint_roughness.jpg: Error opening https://raw.githubusercontent.com/cyberbotics/webots/R2021b/projects/appearances/protos/textures/glossy_car_paint/glossy_car_paint_roughness.jpg
WARNING: Moose (PROTO) > DEF MOOSE_BODY_SHAPE Transform > Shape > GlossyPaint (PROTO) > ImageTexture: Cannot download https://raw.githubusercontent.com/cyberbotics/webots/R2021b/projects/appearances/protos/textures/glossy_car_paint/glossy_car_paint_normal.jpg: Error opening https://raw.githubusercontent.com/cyberbotics/webots/R2021b/projects/appearances/protos/textures/glossy_car_paint/glossy_car_paint_normal.jpg
WARNING: Moose (PROTO) > DEF MOOSE_BODY_SHAPE Transform > Shape > MattePaint (PROTO) > ImageTexture: Cannot download https://raw.githubusercontent.com/cyberbotics/webots/R2021b/projects/appearances/protos/textures/matte_car_paint/matte_car_paint_roughness.jpg: Error opening https://raw.githubusercontent.com/cyberbotics/webots/R2021b/projects/appearances/protos/textures/matte_car_paint/matte_car_paint_roughness.jpg
WARNING: Moose (PROTO) > Transform > HingeJoint > Slot > MooseWheel (PROTO) > Solid > DEF TIRE Shape > GenericTireAppearance (PROTO) > ImageTexture: Cannot download https://raw.githubusercontent.com/cyberbotics/webots/R2021b/projects/vehicles/protos/abstract/textures/tire_texture_base_color.jpg: Error opening https://raw.githubusercontent.com/cyberbotics/webots/R2021b/projects/vehicles/protos/abstract/textures/tire_texture_base_color.jpg
WARNING: Moose (PROTO) > Transform > HingeJoint > Slot > MooseWheel (PROTO) > Solid > DEF TIRE Shape > GenericTireAppearance (PROTO) > ImageTexture: Cannot download https://raw.githubusercontent.com/cyberbotics/webots/R2021b/projects/vehicles/protos/abstract/textures/tire_texture_roughness.jpg: Error opening https://raw.githubusercontent.com/cyberbotics/webots/R2021b/projects/vehicles/protos/abstract/textures/tire_texture_roughness.jpg
WARNING: Moose (PROTO) > Transform > HingeJoint > Slot > MooseWheel (PROTO) > Solid > DEF TIRE Shape > GenericTireAppearance (PROTO) > ImageTexture: Cannot download https://raw.githubusercontent.com/cyberbotics/webots/R2021b/projects/vehicles/protos/abstract/textures/tire_texture_normal.jpg: Error opening https://raw.githubusercontent.com/cyberbotics/webots/R2021b/projects/vehicles/protos/abstract/textures/tire_texture_normal.jpg
WARNING: Moose (PROTO) > Transform > HingeJoint > Slot > MooseWheel (PROTO) > Solid > DEF TIRE Shape > GenericTireAppearance (PROTO) > ImageTexture: Cannot download https://raw.githubusercontent.com/cyberbotics/webots/R2021b/projects/vehicles/protos/abstract/textures/tire_texture_occlusion.jpg: Error opening https://raw.githubusercontent.com/cyberbotics/webots/R2021b/projects/vehicles/protos/abstract/textures/tire_texture_occlusion.jpg
WARNING: CardboardBox (PROTO) > Transform > Shape > PBRAppearance > ImageTexture: Cannot download https://raw.githubusercontent.com/cyberbotics/webots/R2021b/projects/objects/factory/containers/protos/textures/cardboard_box_base_color.jpg: Error opening https://raw.githubusercontent.com/cyberbotics/webots/R2021b/projects/objects/factory/containers/protos/textures/cardboard_box_base_color.jpg
WARNING: CardboardBox (PROTO) > Transform > Shape > PBRAppearance > ImageTexture: Cannot download https://raw.githubusercontent.com/cyberbotics/webots/R2021b/projects/objects/factory/containers/protos/textures/cardboard_box_roughness.jpg: Error opening https://raw.githubusercontent.com/cyberbotics/webots/R2021b/projects/objects/factory/containers/protos/textures/cardboard_box_roughness.jpg
WARNING: CardboardBox (PROTO) > Transform > Shape > PBRAppearance > ImageTexture: Cannot download https://raw.githubusercontent.com/cyberbotics/webots/R2021b/projects/objects/factory/containers/protos/textures/cardboard_box_normal.jpg: Error opening https://raw.githubusercontent.com/cyberbotics/webots/R2021b/projects/objects/factory/containers/protos/textures/cardboard_box_normal.jpg
WARNING: CardboardBox (PROTO) > Transform > Shape > PBRAppearance > ImageTexture: Cannot download https://raw.githubusercontent.com/cyberbotics/webots/R2021b/projects/objects/factory/containers/protos/textures/cardboard_box_occlusion.jpg: Error opening https://raw.githubusercontent.com/cyberbotics/webots/R2021b/projects/objects/factory/containers/protos/textures/cardboard_box_occlusion.jpg
WARNING: Oak (PROTO) > Shape > PBRAppearance > ImageTexture: Cannot download https://raw.githubusercontent.com/cyberbotics/webots/R2021b/projects/objects/trees/protos/textures/oak_leaf.png: Error opening https://raw.githubusercontent.com/cyberbotics/webots/R2021b/projects/objects/trees/protos/textures/oak_leaf.png
WARNING: Oak (PROTO) > Shape > PBRAppearance > ImageTexture: Cannot download https://raw.githubusercontent.com/cyberbotics/webots/R2021b/projects/objects/trees/protos/textures/oak_trunk.jpg: Error opening https://raw.githubusercontent.com/cyberbotics/webots/R2021b/projects/objects/trees/protos/textures/oak_trunk.jpg
WARNING: TexturedBackground (PROTO): Cannot retrieve 'https://raw.githubusercontent.com/cyberbotics/webots/R2021b/projects/default/worlds/textures/cubic/mountains_right.png': Cannot download https://raw.githubusercontent.com/cyberbotics/webots/R2021b/projects/default/worlds/textures/cubic/mountains_r [... cut ...]
WARNING: TexturedBackground (PROTO): Cannot download leftIrradianceUrl: Cannot download https://raw.githubusercontent.com/cyberbotics/webots/R2021b/projects/default/worlds/textures/cubic/mountains_right.hdr: Error opening https://raw.githubusercontent.com/cyberbotics/webots/R2021b/projects/default/worlds/textures/cubic/mountains_right.hdr
INFO: moose_path_following: Starting controller: D:\app\webots\projects\robots\clearpath\moose\controllers\moose_path_following\moose_path_following.exe
You can drive this robot:
Select the 3D window and use cursor keys:
Press 'A' to return to the autopilot mode
Press 'P' to get the robot position
1st target reached

lukicdarkoo, 2021-08-16:
This is related to https://github.com/cyberbotics/webots/issues/3498

omichel, 2021-08-16:
Alternatively, you may also [install Webots from the source](https://github.com/cyberbotics/webots/wiki#installation-of-the-webots-development-environment) to work around this problem.

closed 2021-08-16 by omichel