cyberbotics/webots #3594: LiDAR points distribute as straight line along the coordinate axes

BruceXSK, 2021-08-23:
As shown in the picture
[image]
There are some problems with the LiDAR points.
It seems that they distribute along the x and z axes. For more far away points look more obvious.

This problem appears in the latest version of webots(2021b Release date: July 16, 2021)
But it's ok in the older version (2021b Release date: Apri 28, 2021l), as shown in the picture below:
[image]

Versions:
Webots 2021b Date:July 16, 2021
Ubuntu 20.04

How to fix that? Thanks

omichel, 2021-08-23:
I believe this is consequence of https://github.com/cyberbotics/webots/pull/3230

BruceXSK, 2021-08-24:
> I believe this is consequence of #3230

Thanks

closed 2021-08-24 by BruceXSK

pglira, 2022-05-24:
We are still observing the same issue using Ubuntu 20.04 and Webots 2022a.

[image]

Shouldn't this be fixed? If not, how can we avoid this behaviour?

Thanks for your support.

pglira, 2022-05-25:
We've tried different things in the meanwhile, unfortunately none of it solved the issue.

Currently we are using the Velodyne lidar sensor instead of the generic one, but as you can see here the point cloud is still not correct:

[image]

omichel, 2022-05-25:
This loss of precision was introduced in #3230 to fix a more annoying bug. Feel free to investigate #3230 and propose a patch.

pglira, 2022-05-25:
Thanks for the response. Is there any workaround? E.g. by choosing different settings for the lidar sensor?