cyberbotics/webots #3648: Webots crashes for null radius in `Extrusion`

ad-daniel, 2021-09-07:
**Describe the Bug**
As surfaced here https://github.com/cyberbotics/webots/pull/3627, when changing the radius of the signs (`CautionSign`, `ExitSign`,...) to 0 or to a negative value, it crashes Webots. In practice the problem likely stems from the `Extrusion` sub PROTO.

Although it's good practice to sanitize the parameters, it isn't normal that it crashes regardless.

It is not a regression.

ad-daniel, 2021-10-08:
Originally it was crashing on `feature-enu-plan` branch, however I just tested and it still crashes both in that branch and in master (ubuntu 20).

omichel, 2021-10-08:
I could make it crash on Windows, by pressing the little "down" button several times until the value of radius reaches 0. But if I enter 0 or -1, it doesn't crash and I am getting the same warning as @llessieux.

llessieux, 2021-10-08:
Indeed I can reproduce the crash that way too.
I might have a look at it next week too. Crashes tend to be easy to fix ;)

omichel, 2021-10-10:
Thank you for the detailed report. Yes, you go ahead with a fix, preferably on the master branch. We will be happy to review it.

omichel, 2021-10-11:
> By the way, is there anyway to build Webots using Visual Studio?

I never tried, but in theory, it should be possible.

llessieux, 2021-10-12:
Also why are QPointer not used to prevent that kind of things?
I think that there are made for that.

llessieux, 2021-10-12:
I think that I got it now but I instrumented and modified the code so much that it will take me a while to figure out the minimum set of changes for it.
Basically I think that the editors are keeping references to deleted object due to the regeneration somehow.
Adding connect to the destroyed signal to disable the editor might have done the trick.
But I am certainly not sure that something more fundamental is not broken here. The editors should probably have been cleared and recreated but somehow the event might have been queued.

omichel, 2021-10-12:
> Also why are QPointer not used to prevent that kind of things?

Yes, that could help indeed...

omichel, 2021-10-13:
This is because the node you are modifying (Extrusion) is a PROTO node which runs Javascript templating. Therefore, if you modify a field parameter, the node needs to be regenerated as the change in field parameter may impact outcome of the execution of the Javascript code that generates the node.

llessieux, 2021-10-13:
Understood.