cyberbotics/webots #3655: Make install command issue

RTrioux, 2021-09-08:
**Describe the Bug**
I'm trying to package webots from sources for Arch Linux community.
I have some issues during `make install` command.

1. I don't have any ros distribution installed so  `ros` is not compiled in `webots/projects/default/controllers/ros/` causing a 
"File or folder not found: projects/default/controllers/ros/ros"  later in `generate_projects_files.py`.
2. Beside that the `webots_distro` script looks for `python36`   `python37`   and `python38` folders in `webots/lib/controller/` which don't exist since I only have python 3.9 installed causing an error. The only solution I have found so far is creating empty folders there so that the installation can resume. 

I guess for both cases I can just create some mock-up files/folders but I was wondering if you know a proper way to do that in the installation scripts. 
[image]

[image]

ad-daniel, 2021-09-08:
Hi, did you follow [this](https://github.com/cyberbotics/webots/wiki/Linux-installation) procedure?

RTrioux, 2021-09-08:
I have followed the [wiki](https://github.com/cyberbotics/webots/wiki/Linux-installation) as much as possible since I'm packaging for ArchLinux.
Note that I can compile and use webots without any problem.

omichel, 2021-09-10:
It seems this is because you didn't install the optional dependencies, in particular ROS and Python 3.6. Can you check if the ros controller was built during the build process? Similarly, can you check if the Python 3.6 wrapper was built (in addition to other versions of the Python wrappers)?

RTrioux, 2021-09-13:
Thanks for your answer, 
Yes I would like to package it such a way that a non Ros user can still install webots from sources since these packages are optional [Ros installation may not be straightforward on all plateforms].
Beside Python39 (lastest python version available atm on this distribution), no other Python wrapper are built i.e `lib/controller/python39` is the only one.

Worst case scenario I'll just create dummy files so that the installation can proceed.

lukicdarkoo, 2021-09-14:
> Yes I would like to package it such a way that a non Ros user can still install webots from sources since these packages are optional

I would think of the current behavior as a sanity check. Webots is meant to be shipped with all these dependencies, so if you are missing something then it is useful that `make` reminds about it.

As an alternative, you can make empty folders, e.g. `mkdir lib/controller/python37` to make `make` happy. Maybe we can include this approach in the documentation.

RTrioux, 2021-09-14:
Ok thanks then I'll go with that alternative.
Concerning the documentation I guess I should open another PR.

closed 2021-09-14 by RTrioux