cyberbotics/webots #3656: Incorrect warping of in `Extrusion.proto`

ad-daniel, 2021-09-08:
**Describe the Bug**

1. Load world [extrusion_bug.zip](https://github.com/cyberbotics/webots/files/7131556/extrusion_bug.zip), it looks like:
[image]

2. Change last line of `spine` field from `0 20 0` to `0 20 5`. The extrusion warps as intended.
[image]

3. Set `z` coordinate back to 0.
4. Change last line of `spine` field from `0 20 0` to `5 20 0`. The warping direction is correct, but the short and long edges have been swapped.
[image]

Appears to be the case in 2021b already and therefore other PROTO like `Fence` and `CrashBarrier` are also affected.

joachimhgg, 2021-11-23:
[Extrusion_with_comments.zip](https://github.com/cyberbotics/webots/files/7589355/Extrusion_with_comments.zip)
I commented the proto to verify that we implemented all the conditions based on https://www.web3d.org/documents/specifications/19775-1/V3.3/Part01/components/geometry3D.html.

joachimhgg, 2021-11-24:
[image]
1 = line of `spine` field : `0 20 0`
2 = line of `spine` field : `5 20 0`
3 = line of `spine` field : `0 20 5`

When reproducing the 3 cases in Blender, we got the same behavior.

omichel, 2021-11-25:
I wrote a [debug proto](https://github.com/cyberbotics/webots/files/7601202/extrusion_bug.zip) allowing to display the axes for each spine point of the extrusion and found that the axes are not orthonormal, should be fixed:

[image]

Then, when changing the last spin value from `0 20 0` to `0.0001 20 0`, the axes are swapped, whereas they should remain about the same:
[image]

I would suggest to continue the investigation, based on this debug proto, and display more axes if needed.

omichel, 2021-11-25:
Also when displaying the orientation values for the X-axis, something looks wrong as not all orientations are defined:
```js
if (orientation[i] !== undefined) {
  console.log('orientation[' + i + ']: ' + orientation[i].x + ' ' + orientation[i].y + ' ' + orientation[i].z);
  point = wbrotation.rotateVector3ByRotation(orientation[i], point);
} else
  console.log('orientation[' + i + '] undefined');                
```
gives:
```
INFO: 'ExtrusionShape.proto': JavaScript output: orientation[0]: 0 0 1
INFO: 'ExtrusionShape.proto': JavaScript output: orientation[1]: 0 0.25 0.75
INFO: 'ExtrusionShape.proto': JavaScript output: orientation[2]: 0 0.5 0.5
INFO: 'ExtrusionShape.proto': JavaScript output: orientation[3]: 0 0.75 0.25
INFO: 'ExtrusionShape.proto': JavaScript output: orientation[4] undefined
INFO: 'ExtrusionShape.proto': JavaScript output: orientation[5] undefined
INFO: 'ExtrusionShape.proto': JavaScript output: orientation[6] undefined
INFO: 'ExtrusionShape.proto': JavaScript output: orientation[7] undefined
INFO: 'ExtrusionShape.proto': JavaScript output: orientation[8] undefined
```

PR #3968 (2021-12-02): Fix extrusion
**Description**
It seems like two cases were not implemented or incorrectly done:
- When the first spines were collinear.
- When all the extrusion was collinear.

**Related Issues**
Fix #3665 

**Tasks**
  - [x] Handle case when first spines are collinear.
  - [x] Handle case when all spines are collinear.
  - [x] Comment the proto to ease its readability. (Done by @joachimhgg )
  - [x] Update CrashBarrier.proto
  - [x] Correct orientation behaviour when there is only one orientation define. ` If orientation contain one value, it is applied at all spine points`
  - [x] fix protos [... cut]