cyberbotics/webots #3660: GPS return speed as vector

ShuffleWire, 2021-09-09:
**Is your feature request related to a problem? Please describe.**
Additionally to the speed of on object on which a GPS is located, I wish to know in which direction the object is moving.

**Describe the solution you'd like**
I would like to get the speed vector of the GPS.

**Describe alternatives you've considered**
I currently derivative by hand the position that the GPS return from one step to the other.

Additionally, do common GPS have such capability, or giving only the norm of the speed is due to any physical limitation ?

omichel, 2021-09-10:
I believe we should modify the API, so that the `gps_get_speed` API function returns a 3D vector instead of a scalar value.

omichel, 2021-09-10:
Feel free to go ahead with the implementation, which shouldn't be difficult if you follow our [guidelines](https://github.com/cyberbotics/webots/wiki/Adding-New-Node-and-API-Function#adding-a-new-api-function).

ShuffleWire, 2021-09-10:
Should we modify the gps_get_speed or create a new one, which will keep backward compatibility ?

omichel, 2021-09-10:
I would modify the `gps_get_speed`, breaking backwards compatibility, as this function is not useful and probably very few people are using it. And for those using it, it will be super easy to replace it with the new one. Please target the `develop` branch (not the `master`) to propose your contribution.

lukicdarkoo, 2021-09-10:
> probably very few people are using it.

I am not sure about this, I use it sometimes, it is convenient.

Maybe creating a new one, e.g. `gps_get_velocity_vector` is a better approach.

> do common GPS have such capability

This is a good question. I have never seen a velocity vector output from a GPS device, but I don't see an obvious reason for not having it (we can calculate it from two consecutive positions)

omichel, 2021-09-10:
> Maybe creating a new one, e.g. gps_get_velocity_vector is a better approach.

I agree. It's true that commercial GPSs display velocity as a scalar value on their screen, so the `gps_get_velocity` API function should probably remain as it is.

ShuffleWire, 2021-09-10:
I'm using a very high end GPS (coupled with IMU), which give us vector speed, but I do think that the bare metal GPS on it don't have a physical way to measure speed (it's not a Doppler device, for instance), and I agree with you, derivative consequent position should be enough.
BTW : WbGps.cpp:311 : `mMeasuredSpeed = (mPreviousPosition - t).length() * 1000.0 / mSensor->elapsedTime();`
So currently we derivative position, and we just have to return the same thing, removing the .length()

Additionally: Does the ODE solve could be able to provide such speed like the one surfaced in https://github.com/cyberbotics/webots/issues/3434 ? I'm never confident when derivating stuff, small noise will give wrong output...

lukicdarkoo, 2021-09-10:
> Additionally: Does the ODE solve could be able to provide such speed like the one surfaced in #3434 ? I'm never confident when derivating stuff, small noise will give wrong output...

There are `relativeLinearVelocity` and `relativeAngularVelocity` methods for all solid nodes.

llessieux, 2021-10-02:
I can probably have a go at that if you want ;)

ShuffleWire, 2021-10-03:
Go ahead if you wish, I don't have the energy to work on it now :)
Thank !

llessieux, 2021-10-04:
Ok, I have the code ready ;) But I am not sure how to use it in some sample. Will probably need some guidance for that.
For now, I will update the WbGPS and the gps.h/c bits.

omichel, 2021-10-04:
As an example, I would simply update [this one](https://github.com/cyberbotics/webots/blob/develop/projects/samples/devices/controllers/gps/gps.c) to show how to use the new GPS function.

omichel, 2021-10-04:
Note: you should target the develop branch as this is a new feature, not a bug fix.