cyberbotics/webots #3661: Webots Python Camera recognition Bug

jjing970828, 2021-09-10:
**Describe the Bug**
A clear and concise description of what the bug is.
I'm using Python.
If you look at the third line below, there is an error about WbCameraRecognitionObject*.

 No description of WbCameraRecognitionObject * can be found anywhere in Python's camera-related example.

Perhaps when you call the getRecognitionObject function, the data is output in the form of a swig, which is like a bug that occurred there.

**Steps to Reproduce**
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.
[image]

**System**
 - Operating System: Windows 10
 - Graphics Card: [NVIDIA GeForce gtx 965m]

**Additional context**
Add any other context about the problem here.

stefaniapedrazzi, 2021-09-10:
I'm not sure to understand your code.
Where do you initialize the `CameraObjectRecognition` instance?
Because from your screenshot it doesn't seem to be initialized, so it seems you are calling the `get_id()` function on the `CameraObjectRecognition` class and not on an instance.
If you want to store the function into a variable then should remove the brackets and use simply `id_object = object.get_id`.
otherwise `object` should be a valid instance of `CameraObjectRecognition` and this doesn't seem the case in your code.
 
Could you please check if this minimal code works for you?
 ```
from controller import Robot, Camera

robot = Robot()

# get the time step of the current world.
timestep = int(robot.getBasicTimeStep())
camera = robot.getDevice('camera')
camera.enable(timestep)
camera.recognitionEnable(timestep)

# Main loop:
while robot.step(timestep) != -1:
    num_objects = camera.getRecognitionNumberOfObjects()
    objects = camera.getRecognitionObjects()
    for object in objects:
       id_object = object.get_id()
       print ('id_object ' + str(id_object))
 ```

If yes, then the problem is very likely in the way you built your code.
You should carefully check your classes definition and initialization and adjust it.
Personally I don't see the point of inheriting both from `Camera` and `CameraRecognitionObject` classes given that are completely different object.

For reference, here is the documentation of the `CameraRecognitionObject` class:
http://cyberbotics.com/doc/reference/camera?tab-language=python#camera-recognition-object

jjing970828, 2021-09-10:
I used code you gave me and it's worked! Thanks for help! 
Actually I'm self-studying webots. Its very interesting but also very challenging.
Can you recommend me any guide book or tutorial of Webots API??

 

http://cyberbotics.com/doc/reference/camera?tab-language=python#camera-recognition-object

stefaniapedrazzi, 2021-09-13:
You can find here some tutorials:
https://cyberbotics.com/doc/guide/tutorials

And here is the full reference manual for the Webots API that also contain some examples:
https://cyberbotics.com/doc/reference/nodes-and-api-functions

closed 2021-09-13 by stefaniapedrazzi