cyberbotics/webots #3662: Cannot connect to Webots on pipe ... when too much controllers in world

sean85914, 2021-09-10:
**Describe the Bug**
I'm using Webots to simulate a production line, which there are 52 controllers in the world. When I pressed `Run`, there will be a long delay, then the consoles shown many lines look like
```
Cannot open pipe file: \\.\pipe\webots_2900_1138889167 after trying for 5 seconds
Cannot connect to Webots on pipe: \\.\pipe\webots_2900_1138889167
WARNING: 'utils_controller' controller exited with status: 1.
```
and my simulation can't execute correctly since some core controllers were terminated.  
But when I develop the same world to another computer with better CPU (Xeon), the error didn't occurs.

Full console logs are pasted below
```
INFO: logger_controller: Starting controller: python.exe -u logger_controller.py
INFO: utils_controller: Starting controller: python.exe -u utils_controller.py
INFO: basic_conveyor_belt: Starting controller: python.exe -u basic_conveyor_belt.py
INFO: buffer: Starting controller: python.exe -u buffer.py
INFO: basic_machine: Starting controller: python.exe -u basic_machine.py
INFO: lift_conveyor: Starting controller: python.exe -u lift_conveyor.py 0.3 0
INFO: buffer: Starting controller: python.exe -u buffer.py
INFO: basic_machine: Starting controller: python.exe -u basic_machine.py
INFO: lift_conveyor: Starting controller: python.exe -u lift_conveyor.py 0.3 0
INFO: buffer: Starting controller: python.exe -u buffer.py
INFO: basic_machine: Starting controller: python.exe -u basic_machine.py
INFO: lift_conveyor: Starting controller: python.exe -u lift_conveyor.py 0.3 0
INFO: basic_machine: Starting controller: python.exe -u basic_machine.py
INFO: lift_conveyor: Starting controller: python.exe -u lift_conveyor.py 0.3 0
INFO: basic_machine: Starting controller: python.exe -u basic_machine.py
INFO: lift_conveyor: Starting controller: python.exe -u lift_conveyor.py 0.3 0
INFO: basic_machine: Starting controller: python.exe -u basic_machine.py
INFO: lift_conveyor: Starting controller: python.exe -u lift_conveyor.py 0.3 0
INFO: buffer: Starting controller: python.exe -u buffer.py
INFO: basic_machine: Starting controller: python.exe -u basic_machine.py
INFO: lift_conveyor: Starting controller: python.exe -u lift_conveyor.py 0.3 0
INFO: buffer: Starting controller: python.exe -u buffer.py
INFO: basic_machine: Starting controller: python.exe -u basic_machine.py
INFO: lift_conveyor: Starting controller: python.exe -u lift_conveyor.py 0.3 0
INFO: buffer: Starting controller: python.exe -u buffer.py
INFO: basic_machine: Starting controller: python.exe -u basic_machine.py
INFO: lift_conveyor: Starting controller: python.exe -u lift_conveyor.py 0.3 0
INFO: buffer: Starting controller: python.exe -u buffer.py
INFO: basic_machine: Starting controller: python.exe -u basic_machine.py
INFO: lift_conveyor: Starting controller: python.exe -u lift_conveyor.py 0.3 0
INFO: buffer: Starting controller: python.exe -u buffer.py
INFO: basic_machine: Starting controller: python.exe -u basic_machine.py
INFO: lift_conveyor: Starting controller: python.exe -u lift_conveyor.py 0.3 0
INFO: buffer: Starting controller: python.exe -u buffer.py
INFO: basic_machine: Starting controller: python.exe -u basic_machine.py
INFO: lift_conveyor: Starting controller: python.exe -u lift_conveyor.py 0.3 0
INFO: buffer: Starting controller: python.exe -u buffer.py
INFO: basic_machine: Starting controller: python.exe -u basic_machine.py
INFO: lift_conveyor: Starting controller: python.exe -u lift_conveyor.py 0.3 0
INFO: buffer: Starting controller: python.exe -u buffer.py
INFO: basic_machine: Starting controller: python.exe -u basic_machine.py
INFO: lift_conveyor: Starting controller: python.exe -u lift_conveyor.py 0.3 0
INFO: buffer: Starting controller: python.exe -u buffer.py
INFO: test_machine: Starting controller: python.exe -u test_machine.py
INFO: lift_conveyor: Starting controller: python.exe -u lift_conveyor.py 0.3 0
INFO: buffer: Starting controller: python.exe -u buffer.py
INFO: basic_machine: Starting controller: python.exe -u basic_machine.py
INFO: lift_conveyor: Starting controller: python.exe -u lift_conveyor.py 0.3 0
INFO: feeder: Starting controller: python.exe -u feeder.py
INFO: basic_conveyor_belt: Starting controller: python.exe -u basic_conveyor_belt.py
INFO: recycler: Starting controller: python.exe -u recycler.py
INFO: dashboard: Starting controller: python.exe -u dashboard.py
Cannot open pipe file: \\.\pipe\webots_2900_1138889167 after trying for 5 seconds
Cannot connect to Webots on pipe: \\.\pipe\webots_2900_1138889167
WARNING: 'utils_controller' controller exited with status: 1.
Cannot open pipe file: \\.\pipe\webots_2900_1138889167 after trying for 5 seconds
Cannot connect to Webots on pipe: \\.\pipe\webots_2900_1138889167
WARNING: 'basic_machine' controller exited with status: 1.
[... 423 lines cut ...]
  File "C:\Python37\lib\xmlrpc\client.py", line 1309, in send_content
    connection.endheaders(request_body)
  File "C:\Python37\lib\http\client.py", line 1272, in endheaders
    self._send_output(message_body, encode_chunked=encode_chunked)
  File "C:\Python37\lib\http\client.py", line 1032, in _send_output
    self.send(msg)
  File "C:\Python37\lib\http\client.py", line 972, in send
    self.connect()
  File "C:\Python37\lib\http\client.py", line 944, in connect
    (self.host,self.port), self.timeout, self.source_address)
  File "C:\Python37\lib\socket.py", line 728, in create_connection
    raise err
  File "C:\Python37\lib\socket.py", line 716, in create_connection
    sock.connect(sa)
ConnectionRefusedError: [WinError 10061] 無法連線，因為目標電腦拒絕連線。
WARNING: 'dashboard' controller exited with status: 1.
```
(`util_controller` advertised a RPC server and other controllers will communicate with it to get world configuration)

**System**
 - Operating System: Windows 10
 - CPU: Intel Core i7-9850H
 - Graphics Card: Nvidia Quadro T1000
 - RAM: 16G
 - Webots Version: R2021a

omichel, 2021-09-10:
That could be a system limit. The number of named pipes is limited on some systems. What operating system is running your Xeon machine? Did you change any system setting?

sean85914, 2021-09-10:
`What operating system is running your Xeon machine?` > It runs Ubuntu 18.04.  
`Did you change any system setting?` > The Webots is installed in default setting, I think the system setting is not change.   
What system setting might affect the performance? Any commands to get these values?

omichel, 2021-09-10:
I believe the main problem is a limitation in Windows on the number of named pipes. You should probably run this simulation on Linux only.

closed 2021-09-13 by sean85914