cyberbotics/webots #3671: Integrate the streaming viewer into Webots

lukicdarkoo, 2021-09-13:
At the moment, one needs to open `/resources/web/streaming_viewer/index.html` after running `webots --stream`. Maybe, for Docker users, it would be convenient to just run `webots --stream` and being able to open the simulation at e.g. `http://127.0.0.1:8080` (instead of opening the `index.html`). It is slightly less hustle than searching for the `index.html`, especially easier to document.

llessieux, 2021-10-07:
That should be fairly easy. Detect that the requested url is empty and send the index.html.
Then handle the cases for the other two files required.
I can try to add that if you want.

llessieux, 2021-10-07:
So I got the part serving the files easily but when I connect to the server I am getting an error. 
What am I supposed to type in the ws://localhost:1234 section ?
Just leave it like that and type connect?

llessieux, 2021-10-07:
The change are quite minimal and located only in the WbStreamingServer.cpp file.
Let me know if you want me to create a develop branch for it. You can review it then decide what to do with it ;)