cyberbotics/webots #3672: Inconsistent installation instructions for 20.04

Timple, 2021-09-13:
**Documentation page(s) concerned**
https://cyberbotics.com/doc/guide/system-requirements
https://www.cyberbotics.com/doc/guide/tutorial-8-using-ros
https://www.cyberbotics.com/doc/guide/tutorial-8-using-ros?version=develop#!

**What is the problem with the current documentation?**
It took me quite a while to figure out that only the tarball installation is compatible with ubuntu 20.04 (libboost error). Most documentation seems to suggest all should be working fine:

https://cyberbotics.com/doc/guide/system-requirements:
> Webots is ensured to run on the latest Ubuntu Long Term Support (LTS) release, currently version 20.04

https://www.cyberbotics.com/doc/guide/tutorial-8-using-ros
> These examples were tested with ROS noetic and melodic on Linux. 

https://www.cyberbotics.com/doc/guide/tutorial-8-using-ros?version=develop#!
> The Debian and Ubuntu 20.04 **tarball** packages are compatible with ROS Noetic.

**Describe how you would fix it**
More warnings that ubuntu 20.04 only works with the tarball. Or release 2021c which also should solve the debian packages.

stefaniapedrazzi, 2021-09-13:
> It took me quite a while to figure out that only the tarball installation is compatible with ubuntu 20.04 (libboost error). Most documentation seems to suggest all should be working fine

This is not completely true, on Ubuntu 20.04 you can install Webots using the debian package, the snap package or the tarball package and the all the functionalities except the ROS interaction will work fine.

The only known issues are related to the ROS interaction because packages are built with support of a specific ROS version.
In the installation section of the R2021c version we are already explicitly mentioning that when using ROS some of the installation packages could not be well suited:
https://www.cyberbotics.com/doc/guide/installation-procedure?version=develop#installation-on-linux

However, you are correct that we should port all the additional instructions related to ROS version issues depending on the installation method used also to the R2021b documentation.

Timple, 2021-09-14:
Ah, I was unaware that ROS was the exception, as we always use ROS with webots.

Thanks for the improvements!

closed 2021-09-14 by stefaniapedrazzi