cyberbotics/webots #3692: ros controller crashes when enabĺing lidar R2021b

klausener, 2021-09-16:
**Describe the Bug**
I added a lidar device (tested with Velodyne Puck and Hokuyo UTM-30lx) to my robot. 

When I start the simulation (controller: ros) and enable the lidar device with the help of rosservice the ros-controller crashes.

**Operating System:** Ubuntu Bionic 18.04, ROS Melodic

The problem seems to be connected to https://github.com/cyberbotics/webots/issues/1089 and to the commit #3032. In Version R2021a it doesn't appear.

lukicdarkoo, 2021-09-16:
I believe it is related to https://github.com/cyberbotics/webots/pull/3561

Could you please upgrade Webots to R2021b rev1 to confirm:
https://github.com/cyberbotics/webots/releases

klausener, 2021-09-16:
Thank you very much for the very quick answer! You are right, the problem doesn't appear in R2021b rev1! Thank you!

closed 2021-09-16 by klausener