cyberbotics/webots #3738: Windows 11 with Hyper-V installed detected as VM.

llessieux, 2021-10-02:
When running Webots in Windows 11 with Hyper-V installed, the WbSysInfo::isVirtualMachine will return true even when Webots is running on the Host system.

This cause a rendering performance degradation due to excessive glFlush calls.

**Steps to Reproduce**
Run Webots with gdb on a Windows 11 machine with Hyper-V installed.
put breakpoint in isVirtualMachine and check the result.

**Expected behavior**
Webots correctly detects that it is not running in a VM.

**System**
 - Windows 11 x64 with Hyper-V installed
 - Nvidia RTX 3090 24GB.

Now what is strange is that the 31st bit of the ECX register in the result of the CPUID EAX=1 call is set.
This indicates the presence of an HyperVisor.
See [CPUID](https://en.wikipedia.org/wiki/CPUID#Calling_CPUID) information.

Now this might well be a Windows 11 ONLY issue.
I can't check that on Windows 10.

omichel, 2021-10-04:
I just checked on Windows 10 with a Ryzen 7 4800HS and `WbSysInfo::isVirtualMachine()` returns false as it should. So it may be a Windows 11 ONLY issue.

llessieux, 2021-10-04:
Do you have Hyper-V enabled on that system?

llessieux, 2021-10-04:
I think that it is an Hyper-V issue basically in the code. The way it handles the Hyper-V host is probably not correct.
I will see if I can disable Hyper-V to check which part is the issue.

PR #3744 (2021-10-06): Attempt to fix VM detection on Windows
Might fix #3738. Taken from https://stackoverflow.com/a/66974571/810268
closed 2021-10-06 by omichel