cyberbotics/webots #3853: Downloading assets needs a long time

JiaminGuo, 2021-11-01:
Thanks you so much for reading this issue.

We have been using webots for many years.  Since the release of 2021b, we can’t wait to try the new version. We found this sentence in your webset "For this release we worked hard at finding solutions to reduce the size of the Webots installation packages. To this end, we removed textures, meshes and sounds from the package. These assets are now downloaded from the web only when needed and cached for subsequent use.". However, this improvement is very unfriendly to Chinese users.  When I open a project,  it takes no less than 20 minutes to download assets when I use VPN. It will take longer if I don’t use the VPN.  Could you please publish an 2021b installation package which has textures, meshes and sounds to let the users choose.

ad-daniel, 2021-11-01:
It should only take long the first time. After the first download, the textures are cached and would be re-used. So the difference should only be that before you had to download everything when you installed webots (including stuff you don't use) whereas now you only download what you need and when you need it. After that first download, the same world or any other that uses the same texture will load it from cache.

BenjaminDeleze, 2021-11-15:
I'm closing this for now, feel free to reopen it if you have additional issues.

closed 2021-11-15 by BenjaminDeleze

XavierXYG, 2022-04-10:
Set a proxy in Tools->Preference->Network may be helpful

XavierXYG, 2022-04-10:
Setting a proxy in Tools->Preference->Network might help.