cyberbotics/webots #3863: supervisor does not remove nodes inside proto field MFNode

Thomas-Feldmeier, 2021-11-02:
create a controller, proto and world file from the text below.
run the world, select the tree, press the - key (minus on numeric keypad is possible too).
now the tree is not visible in the graphics window, but it is still visible in the scene tree.
click on tree in the scene tree, press 'del' key to remove it.
webots will crash.

if the proto is simplified to
{
  Group {
    children IS Test
  }
}
then webots will crash after pressing - key.

the bug does not appear if the selected object is not inside of an proto:
click on the proto, convert it to base nodes, no crash happens if the tree is deleted by pressing - key.

webots --sysinfo
System: Linux 5.11.0-38-generic x86_64
Processor: Intel(R) Core(TM) i5-4590 CPU @ 3.30GHz
Number of cores: 4
OpenGL vendor: NVIDIA Corporation
OpenGL renderer: GeForce GTX 1050/PCIe/SSE2
OpenGL version: 4.6.0 NVIDIA 460.91.03

webots --version
Webots version: R2021b

Ubuntu 20.04

TestCrash.cpp:

#include <webots/Supervisor.hpp>
#include <webots/Keyboard.hpp>

#include <stdio.h>

using namespace std;
using namespace webots;

int main(int argc, char **argv) {
  Supervisor *robot = new Supervisor();
  int timeStep = (int)robot->getBasicTimeStep();
  webots::Keyboard *keyboard = robot->getKeyboard();
  keyboard->enable(timeStep);
  while (robot->step(timeStep) != -1) {
    webots::Node* selected = robot->getSelected();
    int key = keyboard->getKey();
    if(key=='-' && selected) {
       selected->remove();
    }
  }
  delete robot;
  return 0;
}

-----------------------------

TestCrash.proto:

#VRML_OBJ R2021b utf8

PROTO TestCrash [
  field MFNode     Test [
  ]
]
{
  Group {
    children [
      Group {
        children IS Test
      }
    ]
  }
}

--------------------------------

TestCrash.wbt:

#VRML_SIM R2021b utf8
WorldInfo {
}
Viewpoint {
}
TexturedBackground {
}
TexturedBackgroundLight {
}
TestCrash {
  Test [
    PottedTree {
    }
  ]
}
Robot {
  controller "TestCrash"
  supervisor TRUE
}

ad-daniel, 2021-11-04:
I'm a bit conflicted about this. If the node is one of the parameters of the PROTO it makes sense to be able to remove it or swap it out, be it manually or through a Supervisor. For instance in case of a `Slot` or any other SFNode, it makes perfect sense. The larger question is what to do about non-visible (i.e not exposed in the PROTO interface). For these specifically I get the feeling that being able to remove them goes against the purpose of it being inside a PROTO to begin with, so perhaps it should be allowed only if the node is exposed.

The crash should be fixed regardless, but I'm wondering how to go about fixing this in the larger sense.

omichel, 2021-11-04:
I agree. When attempting to delete an inner node of a proto, Webots should display a warning in the console and should actually refuse to delete this node. I see no good use case for deleting an inner node inside a proto. If such a node could be deleted, then it should be exposed as field of the proto.

PR #3867 (2021-11-05): Fix node deletion from supervisor not refreshing scene tree
**Description**
Fixes #3863.

Although node deletion using the "delete" button and node deletion from supervisor ultimately rely on a same call to `WbNodeOperations::deleteNode` in order to clear it, when deleting from supervisor the scene tree wasn't refreshed (i.e it didn't trigger `WbMultipleValue::itemRemoved` in `WbTreeItem`). The reason for the difference is that when deleting from keyboard the exposed scene tree node is actively selected [here](https://github.com/cyberbotics/webots/blob/d6a89d8f6ad9c6b8ac2f2dd822247082e2e1219a/ [... cut]
closed 2021-11-05 by ad-daniel