cyberbotics/webots #3873: Improve `village_winter` crossroad textures

ad-daniel, 2021-11-09:
**Describe the Bug**

All the crossroads in `village_winter` have a different texture compared to the roads, which looks kinda odd. Would be nice to something more consistent.

[image]

BenjaminDeleze, 2021-11-10:
It would be quite easy to replace them with `roundabout` protos.
The problem is that the sumo exporter does not have support for `roundabout` right now.  So we would not be able to regenerate the Sumo network.

Thomas-Feldmeier, 2021-11-10:
in village_winter.wbt, change Crossroad appearence to same as in Road
i changed the first Crossroad, it looks nice
[image]
ad

ad-daniel, 2021-11-10:
Whatever change ends up being picked, it should be done in `feature-enu-plan` branch to avoid conflicts when merging

closed 2022-01-13 by joachimhgg