cyberbotics/webots #3879: Contradicting information in the Tutorial and in `Add a new object or import an object` dialog

filiparag, 2021-11-10:
**Describe the Bug**

Contradicting information in the introductory tutorial (guide) and in `Add a new object or import an object` dialog.

In [Tutorial 6](https://cyberbotics.com/doc/guide/tutorial-6-4-wheels-robot#separating-the-robot-in-solid-nodes) there is following text:

> The main structure of a Robot model is **a tree of Solid nodes** linked together. The root node of this tree should be a Robot node.
> [...]
> Having these rules in mind, we can start designing the node hierarchy used to model the robot. The first step is to determine which part of the robot should be modeled as a Solid node.

and in the [mentioned dialog](https://github.com/cyberbotics/webots/blob/master/resources/nodes/Solid.wrl#L1) it states:

> A Solid node can be used to represent objects of the simulated environment that **are not robots** (e.g. obstacles, walls, ground, etc.).

[image]

While following the tutorial, I wasn't sure whether to use Solids or Shapes for Robot's `children` property because the guide says, and the other properties expect Solids, but the dialog says not to use them in Robots.

The dialog text should be changed, so it doesn't cause confusion.

omichel, 2021-11-11:
Thank you for reporting this problem.
I would propose to change [this text](https://github.com/cyberbotics/webots/blob/master/resources/nodes/Solid.wrl#L1) to:
```
# A Solid node can be used to represent objects in the simulated environment (e.g. obstacles, walls, ground, robot parts, etc.).
```
Would that make it clearer for you?

filiparag, 2021-11-11:
> Would that make it clearer for you?

Yeah, I think it would better reflect Solid's usage.

PR #3881 (2021-11-11): Update Solid.wrl
Fixes #3879.
closed 2021-11-11 by omichel