cyberbotics/webots #3884: wrong position of the bounding object with ElevationGrid

joachimhgg, 2021-11-12:
[image]

The boundingObject of the Floor of the world CH_VENS appears to be wrong. It seems to be related to the ElevationGrid.

ad-daniel, 2021-11-12:
The shape in the `boundingObject` relies on the USE of the shape itself, so there is no reason for the two to be different.

It is actually a bug of the elevation grid itself, in fact if you check one of the bottoms corners in this map is correct, however the more the height grows, the more the errors increases.
It can be reproduced even with a much smaller grid. The texture is at the right height, but the `boundingObject` isn't.
[image]

[elevation.zip](https://github.com/cyberbotics/webots/files/7528553/elevation.zip)

ad-daniel, 2021-11-12:
The issue is only visual. Changing the `lineScale` adds an offset to the visual representation of the `boundingObjects` in order to make them more visible, the elevation grid just happens to be a case where this factor goes rogue.

The proper fix of this issue would be to remove the dependency on the linescale for `boundingObjects` as currently it wrongfully represents where the contact surface actually is. To make the edges of the `boundingObjects` more visible the thickness of the line should be increased instead.

stefaniapedrazzi, 2022-06-07:
Task of increasing the line thickness seems related to #2512.