cyberbotics/webots #3916: Downloading webots is unauthorized

VladimirVincan, 2021-11-15:
Hello, 

When I try downloading Webots 2021b using the following commands:

wget -qO- https://cyberbotics.com/Cyberbotics.asc | sudo apt-key add -
sudo apt-add-repository 'deb https://cyberbotics.com/debian/ binary-amd64/'
sudo apt-get update
sudo apt-get install webots

I get the following error:

After this operation, 834 MB of additional disk space will be used.
Do you want to continue? [Y/n] 
Err:1 https://cyberbotics.com/debian binary-amd64/ webots 2021b
  401  Unauthorized [IP: 185.199.109.133 443]
E: Failed to fetch https://objects.githubusercontent.com/github-production-release-asset-2e65be/156228018/305c8797-4ff4-4854-8d6d-35223e165b21?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A/20211115/us-east-1/s3/aws4_request&X-Amz-Date=20211115T144916Z&X-Amz-Expires=300&X-Amz-Signature=ebd298e6591c0c7d9f2d567c67a7fd5f2d1fccda28a393b4e5ae08a3c41323d1&X-Amz-SignedHeaders=host&actor_id=0&key_id=0&repo_id=156228018&response-content-disposition=attachment; filename=webots_2021b_amd64.deb&response-content-type=application/octet-stream  401  Unauthorized [IP: 185.199.109.133 443]
E: Unable to fetch some archives, maybe run apt-get update or try with --fix-missing?

Thanks beforehand!

omichel, 2021-11-16:
Thank you for reporting this problem. I believe I fixed the problem. May I ask you to try again?

omichel, 2021-11-16:
This is apparently caused by a [bug in apt](https://bugs.launchpad.net/ubuntu/+source/apt/+bug/1950095) on Ubuntu 20.04 and was spotted [here](https://github.com/linux-surface/linux-surface/issues/625). Hopefully, it will be fixed soon, so that I can remove my quick fix which consists in disabling the https://cyberbotics.com/debian/binary-amd64/.htaccess file which was redirecting the download to GitHub and uploading `webots_2021b_amd64.deb` in https://cyberbotics.com/debian/binary-amd64/.

VladimirVincan, 2021-11-16:
It works now, thank you very much!

closed 2021-11-16 by omichel