cyberbotics/webots #3917: Huge performance hit when using large ElevationGrid

ShuffleWire, 2021-11-16:
**Describe the Bug**
I'm using an ElevationGrid with large value (about 100 * 100 points). It work well (no issue to display that smoothly), but when I want to change an height, Webots take a will to update. No issue with smaller grid (ie the delay scale with the size of the grid).
I've been initially working with a supervisor to modify in real time the shape of the grid, and found the hard way that it cause crazy delay (if I try to update multiple point at once, Webots could take seconds to complete a step...).

**Steps to Reproduce**
1 - Open : 
<details>
<summary>World</summary>
<pre>#VRML_SIM R2021b utf8
WorldInfo {
  coordinateSystem "NUE"
}
Viewpoint {
  orientation -0.5846802669107566 -0.7637817352556263 -0.27347110702059674 1.6325835190324136
  position -11.007462585955334 19.873638859966295 5.441393012418699
}
TexturedBackground {
}
TexturedBackgroundLight {
}
Shape {
  geometry ElevationGrid {
    height [
      0, 0, 0, 0, 0, 0, 0, 0, 0, 0
    ]
    xDimension 100
    zDimension 100
  }
}
</pre>
</details>
2 - Change the first value of height (from 0 to 1, for instance)

3 - See that Webots take a while to update. (increase grid size if issue not clear enough). During that time one core is used at 100%.

**Expected behavior**
This modification should occur immediately.

**System**
 - Operating System: Debian 11
 - CPU: i9-8950HK
 - Webots master@7e71add8b994ec6f5c0ba3c1d9689f300c778e23

**Additional context**
I've been profiling Webots, hence I provide the output, if anyone interested.
[output.zip](https://github.com/cyberbotics/webots/files/7544734/output.zip)
I've try to get to the heart of the issue (I was initially using a supervisor to get this behaviour, but that not actually needed), but at this point I'm not experienced enough with C++/Qt to dig more into that...

ad-daniel, 2021-11-16:
That's indeed an issue, I've remarked it as well last Friday. Another way of reproducing it is to open `CH_Vens`, right-click the `height` field and select `reset to default`. Webots becomes unresponsive. It succeeds eventually, but it takes a good full minute to do so.

ad-daniel, 2021-11-17:
When changing a single value, the bottleneck is [this](https://github.com/cyberbotics/webots/blob/98722a6068fe0b713df53d93f9936290193d3ae0/src/wren/StaticMesh.cpp#L1892-L1896) step in `StaticMesh::prepareGl`, specifically [this](https://github.com/cyberbotics/webots/blob/98722a6068fe0b713df53d93f9936290193d3ae0/src/wren/StaticMesh.cpp#L1764-L1773) loop in `ComputeTrianglesAndMeshes` is where most of the computation is spent, which also explains why you have the same issue with `IndexedFaceSets` as  well as `ElevationGrid`, so the options are just 2: optimize the loop or don't allow `ElevationGrid` from casting shadows (or at least, not the really big ones).

ShuffleWire, 2021-11-17:
I don't need shadows (at least not when I'm moving points), so I should have tested that before anyway...
I've just tried with and without, and currently, both are very fast, which is concerning as I didn't got this behaviour before (it was laggy)...  
And I confirm setting castShadow to FALSE didn't change anything...

ShuffleWire, 2021-11-17:
TBF, I'm definitely not able to understand what the code is doing... However, even if I remember that we need to resend all the OpenGL infos, it seems that we recompute all the normal and stuff for the whole mesh. Could we not be able to update only the faces for which at least one vertex position have been updated ?

ShuffleWire, 2021-11-17:
I'm tied to use an IndexedFaceSet with only triangle : in my script instead of writing square, I'm writing two triangle (using the same vertexes ofc). So we should bypass all the code that try to make triangle from square on the fly.
however, I've got the same performance issues.
(the new shape is on the top of the two previous in here : [gen_grid_2.zip](https://github.com/cyberbotics/webots/files/7555876/gen_grid_2.zip) )

closed 2023-06-12 by omichel

ShuffleWire, 2023-06-12:
I'm posting here after some testing on my end, and it seems that it improved the situation, but not consistently : 

The test is always the same, given this world : 
```
#VRML_SIM R2023a utf8

EXTERNPROTO "webots://projects/objects/backgrounds/protos/TexturedBackground.proto"
EXTERNPROTO "webots://projects/objects/backgrounds/protos/TexturedBackgroundLight.proto"

WorldInfo {
  coordinateSystem "NUE"
}
Viewpoint {
}
TexturedBackground {
}
TexturedBackgroundLight {
}
Shape {
  geometry ElevationGrid {
    height [
      0, 0, 0, 0, 0, 0, 0, 0, 0, 0
    ]
    xDimension 100
    yDimension 100
  }
}
```
and changing the first height value in the GUI by 1 increment multiples times

Oddly enough, when using a ElevationGrid of 10*10 or 1000*1000, it seems to work just fine, but using a 100*100 make it lag significantly more, with an observable spike of CPU to 100%

Anyway that seems to be a great improvement, and if I need to I could still use bigger grid that I need to to to gain in performance, but it seems odd !

omichel, 2023-06-13:
I believe I found the cause of the problem. The mesh shadow heavy computation is needed for receiving shadows, not casting the shadow of the mesh, this is why the `castShadow` change has no effect on this computation. This is normal.

*Note*: I found nevertheless a bug. When the shadows are globally disabled, we could safely skip this computation, which is currently not the case. I am working on a patch for this.