cyberbotics/webots #3925: Incorrect Python version used when importing controller?

Olegas, 2021-11-16:
**Describe the Bug**
When starting a world with Python controller used, it failed with the following error

```
INFO: referee: Starting controller: /usr/local/Cellar/python@3.9/3.9.8/bin/python3.9 -u referee.py
Traceback (most recent call last):
  File ".../controllers/referee/referee.py", line 14, in <module>
    from controller import Supervisor, AnsiCodes, Node
  File "/Applications/Webots.app/lib/controller/python39/controller.py", line 31, in <module>
    import _controller
ImportError: dlopen(/Applications/Webots.app/lib/controller/python39/_controller.so, 2): Library not loaded: /Library/Frameworks/Python.framework/Versions/3.9/Python
  Referenced from: /Applications/Webots.app/lib/controller/python39/_controller.so
  Reason: image not found
WARNING: 'referee' controller exited with status: 1.
```

I've found in `/Applications/Webots.app/lib/controller` there are folders `python39` and `python39_brew`. So, I guess they are for system's bundled Python and one installed with Homebrew.

If I remove the `python39` folder contents and instead of it put the one from `python39_brew` folder, my world starts successfully without any errors.

In the webots settings python is set to exact location on Homebrew Python 3.9 - /usr/local/Cellar/python@3.9/3.9.8/bin/python3.9

So, webots somehow uses incorrect include paths or something?

**System**
 - macOS BigSur, Python 3.9 from Homebrew

Olegas, 2021-11-16:
I've found the detection code here: https://github.com/cyberbotics/webots/blob/c4508c0c2c9b8d11cdb5d96c1fe093d812e6aa0b/src/webots/control/WbController.cpp#L517-L523

If I execute the command webots are searching for Python path with, it gives me the following result:

```
$ /usr/local/Cellar/python@3.9/3.9.8/bin/python3.9 -c 'import sys; print(sys.exec_prefix)'
/usr/local/opt/python@3.9/Frameworks/Python.framework/Versions/3.9
```

omichel, 2021-11-17:
Do you have any idea on how this could be fixed?

Olegas, 2021-11-17:
What if we compare against `/usr/local/Cellar` not a `sys.exec_prefix` but a `mPythonCommand`.
So, at least one can supply a full path to settings to fix this issue.
But I still think this is not a general solution, sort of a hotfix (

omichel, 2021-11-17:
That seems to be a pretty good solution, as in any case we already have `"/usr/local/Cellar"` hard-coded in there. May I let you go ahead with a PR implementing this fix?

omichel, 2021-11-17:
@Olegas: I just sent you an invite to become a member of the Webots committers so that you don't need to fork this repository to propose a PR.

PR #3931 (2021-11-18): Change Python installation detection for macOS
**Description**
On some systems detecting a Python installation path gives incorrect result. In example:
```
/usr/local/Cellar/python@3.9/3.9.8/bin/python3.9 -c 'import sys; print(sys.exec_prefix)'
/usr/local/opt/python@3.9/Frameworks/Python.framework/Versions/3.9
```

The solution is to not execute Python at all and just compare `mPythonCommand` against `/usr/local/Cellar`. In this case a user can explicitly declare the version they uses by setting it in Webots settings.

**Related Issues**
This fixes #3925
closed 2021-11-18 by Olegas