cyberbotics/webots #3946: Crash with the backward compatiblity with world R2021b open in Webots R2022a `feature-enu-plan`

BenjaminHug8, 2021-11-22:
**Describe the Bug**
If the following world created in `R2021b` is open with the `feature-enu-plan` branch (R2022a), Webots will crash 100% of the time.

**Steps to Reproduce**

1. Download this world [my_world_copy.zip](https://github.com/cyberbotics/webots/files/7581784/my_world_copy.zip)
2. Open it with the Webots `feature-enu-plan` branch (R2022a).
3. Try to open any other world or to create a new one.
4. Webots crashes.

**Additional context**
If the version of the world is changed from `R2021b` to `R2022a`, it will not crash anymore.

lukicdarkoo, 2021-11-22:
It crashes when you run a specific controller or just the world file? The following world opens just fine:
```
#VRML_SIM R2021b utf8
WorldInfo {
}
Viewpoint {
  orientation -0.08897640720692798 0.554154347018585 0.8276449472092201 3.4058220155379852
  position -0.21258049783175004 0.6358608611297873 0.3100127374587124
}
TexturedBackground {
}
TexturedBackgroundLight {
}
CircleArena {
  rotation 1 0 0 1.5708003061004252
}
Robot {
  children [
    HingeJoint {
      jointParameters HingeJointParameters {
        axis 0 1 0
        anchor 0 0 0.025
      }
      device [
        RotationalMotor {
          name "left wheel motor"
        }
      ]
      endPoint Solid {
        translation 0 0.045 0.025
        rotation 1 0 0 0
        children [
          DEF WHEEL Shape {
            appearance PBRAppearance {
              baseColor 1 0 0
              roughness 1
              metalness 0
            }
            geometry Cylinder {
              height 0.01
              radius 0.025
            }
          }
        ]
        name "left wheel"
        boundingObject USE WHEEL
        physics Physics {
        }
      }
    }
    HingeJoint {
      jointParameters HingeJointParameters {
        axis 0 1 0
        anchor 0 0 0.025
      }
      device [
        RotationalMotor {
          name "right wheel motor"
        }
      ]
      endPoint Solid {
        translation 0 -0.045 0.025
        children [
          USE WHEEL
        ]
        name "right wheel"
        boundingObject USE WHEEL
        physics Physics {
        }
      }
    }
    Transform {
      translation 0 0 0.0415
      rotation 1 0 0 1.5708
      children [
        Shape {
          appearance PBRAppearance {
            baseColor 0 0 1
            roughness 1
            metalness 0
          }
          geometry DEF BODY Cylinder {
            height 0.08
            radius 0.045
          }
        }
      ]
    }
    DistanceSensor {
      translation 0.042 0.02 0.063
      rotation 0.9351130202419243 0.25056300542381216 0.25056300542381216 1.63783
      children [
        DEF SENSOR Transform {
          rotation 0.5773489358556708 -0.5773509358554485 -0.5773509358554485 -2.094395307179586
          children [
            Shape {
              appearance PBRAppearance {
                baseColor 1 1 0
                roughness 1
                metalness 0
              }
              geometry Cylinder {
                height 0.004
                radius 0.008
              }
            }
          ]
        }
      ]
      name "ds0"
      lookupTable [
        0 1020 0
        0.05 1020 0
        0.15 0 0
      ]
      numberOfRays 2
      aperture 1
    }
    DistanceSensor {
      translation 0.042 -0.02 0.063
      rotation 0.9351130202419243 -0.25056300542381216 -0.25056300542381216 1.63783
      children [
        USE SENSOR
      ]
      name "ds1"
      lookupTable [
        0 1020 0
        0.05 1020 0
        0.15 0 0
      ]
      numberOfRays 2
      aperture 1
    }
  ]
  boundingObject Transform {
    translation 0 0 0.0415
    rotation 1 0 0 1.5708
    children [
      USE BODY
    ]
  }
  physics Physics {
  }
  controller "<extern>"
}
```

Let me clear the cache and compile again

BenjaminHug8, 2021-11-22:
It crashes when you try to load a different world after the one I gave is loaded.

lukicdarkoo, 2021-11-30:
Fixed in #3950

PR #3950 (2021-11-30): Fix crash when changing a world after the backward compatibility adapta…
Fixes #3946
closed 2021-11-30 by lukicdarkoo