cyberbotics/webots #3963: Backward compatibility loader doesn't work for a simple robot

lukicdarkoo, 2021-11-29:
More details:
https://github.com/cyberbotics/webots/pull/3950#pullrequestreview-814595020

World file:
[my_world_copy.zip](https://github.com/cyberbotics/webots/files/7581784/my_world_copy.zip)

joachimhgg, 2021-12-08:
FYI, I have noticed that `sick.wbt`, `CH_Vens.wbt`, and `complete_appartment.wbt` are crashing too. Also it crashes when opening a 2022a world with 2021b proto inside (at least when I replaced nao.proto by its 2021b version, the 2022a worlds containing this proto are crashing)

BenjaminDeleze, 2021-12-10:
I think it is still making Webots crash. I have two webots installation on my computer. One is on master (R2021b) and one on develop (R2022a). When I tried to open some worlds (for example `nao_room.wbt`, `nao_demo.wbt` or `pr2.wbt`) from my master repository in Webots R2022a, it crashes.

However, if I take those worlds, without making any changes but I just move them to my develop folder, it works fine. So I guess this is maybe related to PROTOs.

PR #4013 (2021-12-09): Avoid applying the backwards compability fix when USE/DEF is used
Fixes #3963 (including the `sick.wbt` crash)

The USE/DEF mechanism makes a huge mess, I am not sure how to fix it. I am afraid of ending up in a rabbit hole and wasting too much time.

I will give it one short try, but I don't have a quick solution for e.g. `DEF Shape > Cylinder`. The idea is to convert it to `Transform > Shape > Cylinder`, convert all USE nodes to ordinary nodes and then convert them, but that doesn't work properly :/
PR #4026 (2021-12-14): Fix backward compability crash for DEF nodes
Fixes #3963

- Non-relevant change. I deleted `if (!WbWorld::instance()->isLoading())` as it seems that doesn't makes sense as there is an assert above.
closed 2021-12-14 by BenjaminDeleze