cyberbotics/webots #3964: segmentation fault when opening a 2021b world in feature-enu-plan branch

joachimhgg, 2021-11-30:
## Description
 When opening this simple world, we get an segmentation fault.

## To reproduce

1. save the world:

```
#VRML_SIM R2021b utf8
WorldInfo {
}
Viewpoint {
  orientation -0.06750834915738668 0.4204490417174527 0.9048012080633624 3.4301283492059724
  position -0.5330653428885737 1.6720406742243783 1.3428920098698358
}
Solid {
    children [
      DEF WHEEL Shape {
        geometry Cylinder {
          height 0.01
          radius 0.025
        }
      }
    ]
    boundingObject USE WHEEL
}
```
2. Open it in Webots 2022a (`feature-enu-plan` branch)
3.  `File/New world` or open another world to get the segmentation fault.

## Investigation
The issue is not happening anymore when:
- removing the `boundingObject `
- replacing `2021b` by `2022a`
- test it in the `develop` branch

ad-daniel, 2021-11-30:
Likely related to this: https://github.com/cyberbotics/webots/pull/3950
Does it still crash in this branch?

joachimhgg, 2021-11-30:
Indeed it is related, this branch fixed it thanks.

closed 2021-11-30 by joachimhgg