cyberbotics/webots #3985: Warning on snap

BenjaminDeleze, 2021-12-07:
**Describe the Bug**
When launching the snap version of Webots I got a lot of warning like 
```
Fontconfig warning: "/etc/fonts/conf.d/10-hinting-slight.conf", line 4: unknown element "its:rules"
Fontconfig warning: "/etc/fonts/conf.d/10-hinting-slight.conf", line 5: unknown element "its:translateRule"
Fontconfig error: "/etc/fonts/conf.d/10-hinting-slight.conf", line 5: invalid attribute 'translate'
Fontconfig error: "/etc/fonts/conf.d/10-hinting-slight.conf", line 5: invalid attribute 'selector'
Fontconfig error: "/etc/fonts/conf.d/10-hinting-slight.conf", line 6: invalid attribute 'xmlns:its'
Fontconfig error: "/etc/fonts/conf.d/10-hinting-slight.conf", line 6: invalid attribute 'version'
```
and so on.

Webots seems to launch normally after that but these warnings were not there before.

omichel, 2021-12-07:
Did you try [this](https://github.com/cyberbotics/webots/issues/2435#issuecomment-725060754])?

omichel, 2021-12-07:
Also, it seems we not the only ones experiencing this on snap: https://itectec.com/ubuntu/ubuntu-snap-issue-with-fontsconfig-on-ubuntu-20-04/

The problem is documented [here](https://cyberbotics.com/doc/guide/installation-procedure#troubleshooting). I am afraid we cannot do much in Webots about it, except improve the documentation if the workaround doesn't work and we have a better one.

BenjaminDeleze, 2021-12-07:
> Did you try [this](https://github.com/cyberbotics/webots/issues/2435#issuecomment-725060754%5D)?

Yes it did not change anything

omichel, 2021-12-08:
@BenjaminDeleze: May I ask you to test the solution proposed [here](https://itectec.com/ubuntu/ubuntu-snap-issue-with-fontsconfig-on-ubuntu-20-04/)? If that works, I will add it to the documentation.

BenjaminDeleze, 2021-12-08:
It did not change anything

omichel, 2021-12-08:
It seems that [updating the snap from core18 to core20 would fix the problem](https://github.com/octave-snap/octave-snap/issues/67#issuecomment-821853801). But if we do this, I fear it won't work any more on Ubuntu 18.04.

closed 2021-12-08 by omichel