cyberbotics/webots #4003: Incorrect texture mapping of the salamander

BenjaminDeleze, 2021-12-08:
The texture mapping of the salamander in `animated_skin.wbt` is different

BenjaminDeleze, 2021-12-08:
seems to have been broken somewhere in https://github.com/cyberbotics/webots/pull/3566

BenjaminDeleze, 2021-12-08:
The bad mapping have appeared during the translation from ogre format to .fbx.

It is not critical because the salamander still have a texture, but it is less pretty (especially around the head). As I don't know how to fix it (I tried manually with blender and tried to find a ogre exporter), I will let it for later

stefaniapedrazzi, 2022-05-02:
The different mapping was made on purpose, so the fact that it looks different is not an "issue" but the issue is if it looks bad.
What do you mean by "bad mapping"?
Could you point exactly which part has a bad mapping?
We should aim at having a nice looking mapping but not exactly the same as before.
In any case, I can rework it and make it looks better.

It is very complicated to have the exact same old Ogre mapping as before and it is much better to create a new one with Blender as it was done in https://github.com/cyberbotics/webots/pull/3566 (and also when creating the Ogre mapping).
To way to fix is to manually adjust the map in Blender.

BenjaminDeleze, 2022-05-02:
I think that the mapping around the head looks strange. I said "bad mapping", because I had the impression that the previous mapping had a "specific mapping" for the head (small scales that radiate and get bigger) whereas now it seems weird with big scales.

Before:
[image]
Now
[image]

But I am not sure if it is worth to rework it as it looks okay as long as we do not zoom in too much.

closed 2022-06-08 by stefaniapedrazzi