cyberbotics/webots #4017: robot endpoint: pen could not draw at web streaming mode

hanm2019, 2021-12-10:
**Describe the Bug**
pen could not draw at web streaming mode

**Steps to Reproduce**
1. run the `robot/abb/irb/inverse_kinematics.wbt` at web streaming mode

**Expected behavior**
A circle on the paper

**Screenshots**
Screenshots from web:
[image]

Screenshots from webots:
[image]

**System**
 - Operating System: Linux Ubuntu 18.04
 - Graphics Card: NVIDIA GeForce RTX 1080 11 GB
 - Webots Version: 2021a

hanm2019, 2021-12-10:
I did not install `opencv-python`, Did it matter?

BenjaminDeleze, 2021-12-10:
Unfortunately, this feature is not yet supported in web streaming mode/ web animation

closed 2021-12-10 by hanm2019