cyberbotics/webots #4034: Scale parameter should be removed from Robot node

ad-daniel, 2021-12-15:
**Describe the Bug**

The `Robot` node inherits the scale parameter from the `Transform` node, using this parameter however has several issues/bugs:
- If a user sets the density in the `Physics`, it will scale but if it sets the mass, it doesn't.
- If the robot has sensors, like a distance sensor. After scaling it won't work well anymore. (detection point is incorrect, lookup table isn't respected, ...)
- Although it might be possible to "fix" all sensors to work with a smaller uniform scale, for non-uniform ones such as  `scale 0.5 1.5 0.3`, it would be incredibly hard to have the desired behavior for every sensor type

omichel, 2021-12-15:
I believe the `scale` parameter should be allowed only in the `Transform` node and in no other node, as it will cause various problems. Even in a simple `Solid` node, there is a problem if the `mass` is specified in the `Physics` node of the `Solid` as its value won't scale or won't be correct. Also, for the same reason, `Transform` nodes shouldn't contain any `Solid` node (or descendant). When needed, users should use an upper `Solid` node (instead of a Transform) to encapsulate other `Solid` nodes.

omichel, 2021-12-15:
An idea to make things clean would be to introduce a `Pose` node that would be basically like a `Transform`, but without the `scale` field. The `Solid` nodes would then inherit from `Pose` and a `Pose` node would be allowed to include solid children.

closed 2023-06-12 by BenjaminDeleze