cyberbotics/webots #4041: Modification of default worlds prompt is missing when installing from deb

ad-daniel, 2021-12-16:
**Describe the Bug**

**Steps to Reproduce**
1. Download and install a nightly version of 2022a from **deb**
2. Open a random world
3. Zoom-out or make another change
4. Save

Normally it should prompt a message stating that you cannot modify a default project and propose to copy it elsewhere. In practice nothing happens.
Similarly, if you attempt to clean or recompile an existing controller, no prompt is shown and the commands result in errors in the console.

I've tried R2022a, R2021b and R2021a  from deb and they all behave in the same manner, so it isn't a regression but it's an undesired behavior nonetheless.

BenjaminDeleze, 2021-12-16:
I think it is a regression, because I remember testing that for the previous release and it was working. However, I think it is more a problem due to the computer, either a library or a environment variable, something like that.

ad-daniel, 2021-12-16:
By doing `unset WEBOTS_ALLOW_MODIFY_INSTALLATION` the message does appear. It isn't sufficient to set the variable to 0.

PR #4046 (2021-12-17): Fixes behavior when attempting to save in installation folder
Fixes #4041.
closed 2021-12-17 by omichel