cyberbotics/webots #4080: Simulation view cannot display correctly

chenlin41204050, 2021-12-24:
**Describe the Bug**
A clear and concise description of what the bug is.

**Steps to Reproduce**
1. Create a new project from the Wizards menu by selecting the New Project Directory
2. Simulation view cannot display correctly.

**Expected behavior**

**Screenshots**
[image]

**System**
 - Operating System: Windows 10
 - Graphics Card: NVIDIA GeForce GTX 1660 6 GB

**Additional context**
WARNING: RectangleArena (PROTO) > Parquetry (PROTO) > ImageTexture: Cannot download https://raw.githubusercontent.com/cyberbotics/webots/R2022a/projects/appearances/protos/textures/parquetry/chequered_parquetry_base_color.jpg: Error opening https://raw.githubusercontent.com/cyberbotics/webots/R2022a/projects/appearances/protos/textures/parquetry/chequered_parquetry_base_color.jpg
WARNING: RectangleArena (PROTO) > Parquetry (PROTO) > ImageTexture: Cannot download https://raw.githubusercontent.com/cyberbotics/webots/R2022a/projects/appearances/protos/textures/parquetry/chequered_parquetry_roughness.jpg: Error opening https://raw.githubusercontent.com/cyberbotics/webots/R2022a/projects/appearances/protos/textures/parquetry/chequered_parquetry_roughness.jpg
WARNING: RectangleArena (PROTO) > Parquetry (PROTO) > ImageTexture: Cannot download https://raw.githubusercontent.com/cyberbotics/webots/R2022a/projects/appearances/protos/textures/parquetry/chequered_parquetry_normal.jpg: Error opening https://raw.githubusercontent.com/cyberbotics/webots/R2022a/projects/appearances/protos/textures/parquetry/chequered_parquetry_normal.jpg
WARNING: RectangleArena (PROTO) > Parquetry (PROTO) > ImageTexture: Cannot download https://raw.githubusercontent.com/cyberbotics/webots/R2022a/projects/appearances/protos/textures/parquetry/chequered_parquetry_occlusion.jpg: Error opening https://raw.githubusercontent.com/cyberbotics/webots/R2022a/projects/appearances/protos/textures/parquetry/chequered_parquetry_occlusion.jpg
WARNING: RectangleArena (PROTO) > BrushedAluminium (PROTO) > ImageTexture: Cannot download https://raw.githubusercontent.com/cyberbotics/webots/R2022a/projects/appearances/protos/textures/brushed_aluminium/brushed_aluminium_base_color.jpg: Error opening https://raw.githubusercontent.com/cyberbotics/webots/R2022a/projects/appearances/protos/textures/brushed_aluminium/brushed_aluminium_base_color.jpg
WARNING: RectangleArena (PROTO) > BrushedAluminium (PROTO) > ImageTexture: Cannot download https://raw.githubusercontent.com/cyberbotics/webots/R2022a/projects/appearances/protos/textures/brushed_aluminium/brushed_aluminium_roughness.jpg: Error opening https://raw.githubusercontent.com/cyberbotics/webots/R2022a/projects/appearances/protos/textures/brushed_aluminium/brushed_aluminium_roughness.jpg
WARNING: RectangleArena (PROTO) > BrushedAluminium (PROTO) > ImageTexture: Cannot download https://raw.githubusercontent.com/cyberbotics/webots/R2022a/projects/appearances/protos/textures/brushed_aluminium/brushed_aluminium_normal.jpg: Error opening https://raw.githubusercontent.com/cyberbotics/webots/R2022a/projects/appearances/protos/textures/brushed_aluminium/brushed_aluminium_normal.jpg
WARNING: TexturedBackground (PROTO): Cannot retrieve 'https://raw.githubusercontent.com/cyberbotics/webots/R2022a/projects/default/worlds/textures/cubic/mountains_back.png': Cannot download https://raw.githubusercontent.com/cyberbotics/webots/R2022a/projects/default/worlds/textures/cubic/mountains_back.png: Error opening https://raw.githubusercontent.com/cyberbotics/webots/R2022a/projects/default/worlds/textures/cubic/mountains_back.png
WARNING: TexturedBackground (PROTO): Cannot download leftIrradianceUrl: Cannot download https://raw.githubusercontent.com/cyberbotics/webots/R2022a/projects/default/worlds/textures/cubic/mountains_back.hdr: Error opening https://raw.githubusercontent.com/cyberbotics/webots/R2022a/projects/default/worlds/textures/cubic/mountains_back.hdr

ad-daniel, 2021-12-26:
The links you provided appear to be working, is the issue still happening? Are you behind a firewall/VPN? Might have been a temporary problem on github side.

TheChlio, 2022-03-09:
r2021a works

closed 2022-06-02 by stefaniapedrazzi

campagnani, 2022-06-16:
I have the same problem in the computer of my university.

The crasy thig is: some user happens this problem, in the same computer, same system, some users don't have this problems.

Distribuition: Ubuntu 22.04

oktaspc, 2022-06-18:
i have the same issue in my pc

i installed webot using terminal.

Distribuition: Ubuntu 22.04

omichel, 2022-06-18:
Could you please with the latest [nightly build](https://github.com/cyberbotics/webots/releases) of Webots R2022b? It should work with this version.

campagnani, 2022-06-19:
@oktaspc I tested the R2022b and this solved the bug.