cyberbotics/webots #4112: Running Demonstration Controller on Apple M1 Pro Chip and Python 3.8

hydenp, 2022-01-13:
**Describe the Bug**
I am working through the Webots first tutorial [here](https://cyberbotics.com/doc/guide/tutorial-1-your-first-simulation-in-webots) and running into an issue on Hands-On #9 when running on my Mac with M1 Pro and Python 3.8. I am using webots R2021b and have also tried with R2022a

When I try to run the simulation, the following error is in the console: 

```
Traceback (most recent call last):
  File "epuck_go_forward.py", line 5, in <module>
    from controller import Robot, Motor
  File "/Applications/Webots.app/lib/controller/python38/controller.py", line 31, in <module>
    import _controller
ImportError: dlopen(/Applications/Webots.app/lib/controller/python38/_controller.so, 0x0002): tried: '/Users/hydenpolikoff/Webots/my_first_simulation/controllers/epuck_go_forward/_controller.so' (no such file), '/Applications/Webots.app/lib/controller/_controller.so' (no such file), '/Applications/Webots.app/lib/controller/python38/_controller.so' (mach-o file, but is an incompatible architecture (have 'x86_64', need 'arm64e')), '/usr/local/lib/_controller.so' (no such file), '/usr/lib/_controller.so' (no such file)
WARNING: 'epuck_go_forward' controller exited with status: 1.
```

I installed python 3.8 via Homebrew and updated the path in Webot settings to 
`/opt/homebrew/opt/python@3.8/bin/python3`

**Steps to Reproduce**
1.  Following the tutorial [here](https://cyberbotics.com/doc/guide/tutorial-1-your-first-simulation-in-webots) using M1 Pro device

**Expected behavior**
I expected the tutorial to work as expected

**System**
 - Operating System: [macOS Mojave 12.1 with M1 Pro]

**Additional context**
I am not sure about the status of Webots and compatibility with Apple Silicon so if there is something I am missing, please let me know.

omichel, 2022-01-13:
Webots currently runs as a x86_64 binary on M1 Macs through Rosetta 2, which is fine. It also provides python libraries for x86_64l, but not (yet) for M1 processors, so you should install the x86_64 version of python via Homebrew. Explanations on how to do this are provided [here](https://medium.com/mkdir-awesome/how-to-install-x86-64-homebrew-packages-on-apple-m1-macbook-54ba295230f). Basically, it's a matter of typing something like that `PATH=$HOME/bin:/usr/local/bin:$PATH arch -x86_64 /usr/local/homebrew/bin/brew install python@3.8`. Let me know if that works for you.

We do have plans to provide a M1 native version of Webots in the next major release of Webots, including backwards compatibility with non-M1 Macs (with fat binaries).

hydenp, 2022-01-13:
Hi @omichel

Thanks for your quick response! I've done as you suggested but still ran into some issues. However, I was able to solve so I will post here in case others encounter the same.

I installed python3.9 via homebrew using Rosetta 2 Homebrew with x86 prefix etc. I encountered the following error
```
Traceback (most recent call last):
  File "/Users/hydenpolikoff/Webots/my_first_simulation/controllers/epuck_go_forward/epuck_go_forward.py", line 5, in <module>
    from controller import Robot, Motor
  File "/Applications/Webots.app/lib/controller/python39/controller.py", line 31, in <module>
    import _controller
ImportError: dlopen(/Applications/Webots.app/lib/controller/python39/_controller.so, 0x0002): Library not loaded: /Library/Frameworks/Python.framework/Versions/3.9/Python
  Referenced from: /Applications/Webots.app/lib/controller/python39/_controller.so
  Reason: tried: '/Library/Frameworks/Python.framework/Versions/3.9/Python' (no such file), '/System/Library/Frameworks/Python.framework/Versions/3.9/Python' (no such file)
WARNING: 'epuck_go_forward' controller exited with status: 1.
```

It seemed as though something was having trouble recognizing where my python installation was. I had the path in webots set to `usr/local/bin/python` which worked to open the python console in the terminal so I new it was installed correctly.

There is probably a way to fix this better than I did, but I proceeded with the following solution.

I know downloading with dmg and installer from python.org will install to the `Library/Frameworks` location which is where the webots seemed to be looking even though I'd specified the path—so that's what I did. Uninstalled my Rosseta 2 brew installed python3.9 installation. Then went to [python3.9](https://www.python.org/downloads/release/python-399/) release and downloaded and installed the intel-only version of python 3 `macOS 64-bit Intel-only installer` to assure I had the x86 version assuming it would work via Rosseta 2.

I ran the installer and changed my Python command back to simply python3 in the Webots settings and things seem to be working as expected! Hope this could help someone else in the future.

P.S. excited to see native M1 Support coming soon

closed 2022-01-13 by hydenp