cyberbotics/webots #4127: Get an empty array of object with "recognition_objects" topic

BenjaminHug8, 2022-01-18:
Sorry. I always get an empty array of object when using the topic "recognition_objects".  I'm wondering what could be the reason.
[image]
And this is my code:
`void camera_topCallback(const webots_ros::RecognitionObjects::ConstPtr &msg)
{
    camera_topRawvalue = *msg;
    ROS_INFO("Update camera_topRawvalue: numberofobj %d", msg->objects.size());
    //ROS_INFO("FIRST OBJ: %s", msg->objects[0].model.c_str());
}`

_Originally posted by @Liuxingyuming in https://github.com/cyberbotics/webots/issues/3234#issuecomment-1015515193_

BenjaminHug8, 2022-01-18:
@Liuxingyuming I propose to continue to discuss about your issue here instead of inside a close pull request.

Liuxingyuming, 2022-01-19:
Okk, thank u!

Liuxingyuming, 2022-01-19:
I found that, when I echo the topic, the seq of the message is always 0. It seems that the message is not getting updated ？

BenjaminHug8, 2022-01-19:
Would it be possible to upload your package here so I can made some test on it?

Liuxingyuming, 2022-01-19:
I upload my project on github:

https://github.com/Liuxingyuming/Supermarket-Robot-Based-on-webots_ros

The ros code is in /src/webots_demo/src. Thanks.

BenjaminHug8, 2022-01-19:
I got the same issue locally, it seems that the message is not properly built.

epcm, 2022-01-19:
> I got the same issue locally, it seems that the message is not properly built.

So is there any temporary solution for this issue ?

BenjaminHug8, 2022-01-19:
I am working on it, once it will be ready you should be able to build Webots from sources or wait for a nightly build.

BenjaminHug8, 2022-01-20:
The fix has been merged in `master` branch of Webots.

You have now the choice between building Webots from sources or waiting for the next [nightly build](https://github.com/cyberbotics/webots/releases) in order to benefit of this fix.

I will close this issue as it should be solved now, but feel free to reopen it in case there is still something not working :)

closed 2022-01-20 by BenjaminHug8