cyberbotics/webots #4128: Python 3.10 support

SGSSGene, 2022-01-18:
Currently, webots only works with python 3.9 (or lower). It would be nice to have python 3.10 support.

SGSSGene, 2022-01-19:
Very cool, thank you. For reference, I have patched a few things, so I can run the Robocup-HumanoidTC branch.
My changes are too chaotic for a PR, but maybe helpful for the next person working on it.
https://github.com/SGSSGene/webots/commit/dcee03c99bb785fa4cb961e0f738875c40fff91b

omichel, 2022-01-19:
That looks like a very good start. Could you please go ahead and create a draft PR which we will review and possibly complete? I just sent you an invite to become a member of the Webots committers, this should facilitate your contribution.

omichel, 2022-01-20:
Note: @Justin-Fisher is currently developing a new Python API which should be fully independent from the Python version (for Python >= 3.5). See details at #3801. So, it may not be necessary in the long term to maintain manual support for each version of python...

PR #4474 (2022-04-26): changes to compile with python3.10
**Description**
These changes are required to get webots working with Python3.10. The change from ``[0-9]`` to ``[0-9]*`` in the sed command is necessary to set ``PYTHON_VERSION`` to ``3.10`` and not ``3.1`` like it currently does.

While I see https://github.com/cyberbotics/webots/pull/3801 is in progress, this is necessary for a short term solution for Ubuntu Jammy since it uses Python3.10 by default.

**Related Issues**
Resolves: #4128

**Additional Notes**

``src/webots/control/WbLanguageTools.cpp`` contains a lot of references to python versions, but I h [... cut]
closed 2022-04-26 by ijnek