cyberbotics/webots #4129: small bug in streaming viewer wwi/Server.js

mosyang, 2022-01-19:
**Describe the Bug**
**Path:**  resources/web/wwi/Server.js
Inside onMessage(event) function, there should be a judgement before accessing this._view.x3dScene
Otherwise, streaming headless mjpeg will get an exception on it.

Current:
```
this._view.x3dScene.prefix = this._httpServerUrl + '/';
```
Expected:  
```
if (typeof this._view.x3dScene !== 'undefined')
    this._view.x3dScene.prefix = this._httpServerUrl + '/';
```

BenjaminDeleze, 2022-01-19:
What do you mean exactly be `headless` ?

mosyang, 2022-01-20:
@BenjaminDeleze  I launch simulation server and session server inside docker container without connecting to X-server or screen UI.  That means, webots run with virtual screen xvfb (which means headless) on remote VM.  When I used streaming viewer to connect session server with mjpeg mode, exception happens in Server.js    
https://cyberbotics.com/doc/guide/web-simulation#using-docker
`ENV DISPLAY=:99`
`Xvfb :99 -screen 0 1024x768x16 &`

After some investigation, I solve this exception by adding `this._view.x3dScene` judgement like the first description.

BenjaminDeleze, 2022-01-20:
Thank you, I can reproduce it. It will be fixed in #4136

PR #4136 (2022-01-20): Fix check in WbServer.js
**Description**
Fix #4129
closed 2022-01-20 by BenjaminDeleze