cyberbotics/webots #4131: Track vanishes after some seconds if not inside an big enough bounding box

Thomas-Feldmeier, 2022-01-19:
Webots 2022a, Ubuntu 20.04, Sample World samples > devices > track.wbt

run world, fast mode, all is working well

stop and reload world (world is now at 0:00:00:000, not running)
DEF CONVEYOR_BELT Robot > children > Shape > geometry Box > size
change from 0.66 0.1 0.12 to 0.3 0.1 0.12

fast mode, track will vanish at about 17 seconds simulation time:
Track rotation -inf inf nan 0
After click on rotation field and another place the rotation is reset to 0 1 0 0 and track is visible again

stop and reload world, change as above, delete BOX_3 BOX_2 BOX_1 (red boxes)

run world in fast mode, all is working well (without boxes on top, there is no rotational force)

omichel, 2022-01-19:
I cannot reproduce this problem on Windows with Webots R2022a.

ad-daniel, 2022-01-19:
I can reproduce it on Ubuntu 20, definitely a bug

closed 2022-01-20 by ad-daniel