cyberbotics/webots #4134: cannot import _controller.so webots 2022a

Jenifen, 2022-01-19:
Error occured while i run the Simulation :  
ImportError: /usr/local/webots/lib/controller/python38/_controller.so: undefined symbol: _ZNK6webots3GPS14getSpeedVectorEvpen  

[image]

**System**
 - Operating System: Linux Ubuntu 20.04
 - Graphics Card: NVIDIA GeForce GTX 1650
 - webots 2022a
 - python38

Thanks !

ad-daniel, 2022-01-19:
Can you post the values of:
```
echo $WEBOTS_HOME
echo $PYTHONPATH
```
They should reference the 2022a folder

Jenifen, 2022-01-19:
thanks for your response : 

**$WEBOTS_HOME** : /usr/local/webots
**$PYTHONPATH** : /opt/ros/foxy/lib/python3.8/site-packages

closed 2022-01-19 by Jenifen