cyberbotics/webots #4146: Web interface: Problems with touch devices

bjost2s, 2022-01-24:
When using the webots simulation e. g. [here](https://cyberbotics1.epfl.ch/open-roberta/), touch events are not evaluated as expected.
- Moving the scene with one finger leads to disappearing of the whole scene
- Resizing with two fingers does not work. 

The only way to get the scene back is reconnecting.

**System**
 - Operating System: Android 7 (higher versions do not work either)
 - Browser: Chrome

bjost2s, 2022-01-24:
Thanks for taking care!

BenjaminDeleze, 2022-01-25:
The fix has been merged.
You can test it here: https://www.cyberbotics.com/animations/soccer_example/soccer.html

If you want to backport it to your installation, please refer to the changes in https://github.com/cyberbotics/webots/blob/master/resources/web/wwi/MouseEvents.js

PR #4149 (2022-01-25): Fix mobile navigation
**Description**
Fix #4146 

I restored it close to what it is robotbenchmark, you can test it there: https://www.cyberbotics.com/animations/soccer_example/soccer.html

I advise you to switch in fullscreen mode to test on mobile.