cyberbotics/webots #4151: cannot import module Supervisor or Robot from controller

METEORITENMAX, 2022-01-25:
When i start the simulation i get the error: 

```
$ ros2 launch webots_spot spot_launch.py 
[INFO] [launch]: All log files can be found below /home/max/.ros/log/2022-01-25-14-44-19-131791-max-MS-7923-25113
[INFO] [launch]: Default logging verbosity is set to INFO
WARNING: Target Webots version `R2021b` is not found, fallback to `R2022a`
[INFO] [webots-1]: process started with pid [25116]
[INFO] [driver-2]: process started with pid [25118]
[INFO] [robot_state_publisher-3]: process started with pid [25120]
[driver-2] Cannot open file: /tmp/webots-25116-34dnQj/WEBOTS_SERVER (retry count 1)
[robot_state_publisher-3] Parsing robot urdf xml string.
[robot_state_publisher-3] [INFO] [1643118259.249444207] [robot_state_publisher]: got segment 
[driver-2] Cannot open file: /tmp/webots-25116-34dnQj/WEBOTS_SERVER (retry count 2)
[driver-2] Cannot open file: /tmp/webots-25116-34dnQj/WEBOTS_SERVER (retry count 3)
[driver-2] Cannot open file: /tmp/webots-25116-34dnQj/WEBOTS_SERVER (retry count 4)
[driver-2] Cannot open file: /tmp/webots-25116-34dnQj/WEBOTS_SERVER (retry count 5)
[driver-2] Cannot open file: /tmp/webots-25116-34dnQj/WEBOTS_SERVER (retry count 6)
[driver-2] Traceback (most recent call last):
[driver-2]   File "/home/max/webots_ws/install/webots_spot/lib/python3.8/site-packages/webots_spot/spot_driver.py", line 8, in <module>
[driver-2]     from controller import Supervisor
[driver-2]   File "/usr/local/webots/lib/controller/python38/controller.py", line 31, in <module>
[driver-2]     import _controller
[driver-2] ImportError: /usr/local/webots/lib/controller/python38/_controller.so: undefined symbol: _ZNK6webots3GPS14getSpeedVectorEv
[driver-2] terminate called after throwing an instance of 'std::runtime_error'
[driver-2]   what():  The webots_spot.spot_driver.SpotDriver plugin cannot be found (C++ or Python).
[ERROR] [driver-2]: process has died [pid 25118, exit code -6, cmd '/home/max/webots_ws/install/webots_ros2_driver/lib/webots_ros2_driver/driver --ros-args --params-file /tmp/launch_params_utx2jmv9'].
```
I want a Supervisor inside my ros node to access a boundingObject, but i cant import the Supervisor module, because of `_controller`. It is the same error as here: #4134 

The error was solve by exporting `LD_LIBRARY_PATH`.
I also have those environment variables exported:
```
export WEBOTS_HOME=/usr/local/webots
export PYTHONIOENCODING=UTF-8
export PYTHONPATH=$PYTHONPATH:$WEBOTS_HOME/lib/controller/python38
export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:$WEBOTS_HOME/lib/controller
```

My `resource/spot.urdf` looks like this:
```
<?xml version="1.0" ?>
<robot name="Spot">
    <webots>
        <plugin type="webots_spot.spot_driver.SpotDriver" />
    </webots>
</robot>
```
Do i need a plugin for the supervisor?
Or is this warning a problem:
```
WARNING: Target Webots version `R2021b` is not found, fallback to `R2022a`
```
I want to use R2022a, i don't know where to set the target version.

**System**
 - Operating System: Linux Ubuntu 20.04
 - Graphics Card: NVIDIA GeForce GTX 970 4GB
 - Webots R2022a
 - ROS 2 foxy
 - python 3.8.10

BenjaminHug8, 2022-01-25:
@METEORITENMAX I have just checked and it seems we have fixed an issue in `webots_ros2` with the Supervisor API after the release of `R2022a`.

What I would suggest is that you install a [nightly](https://github.com/cyberbotics/webots/releases) version of Webots and update your version of `webots_ros2` to the `1.2.2` version.

Be sure you can at least launch the mavic example as it would confirm you do not have any problem with your Python and Webots.

Let me know it that works for you and do ask if you have any question on this!

METEORITENMAX, 2022-01-25:
Thanks that helped!!
The mavic didn't moved and terminal throw an error. There were modified files in `webots_ros2_driver/webots`. 

I installed the nightly build and updated `webots_ros2` to `1.2.2`
Got this output:
```$ git status
HEAD detached at 1.2.2
Changes not staged for commit:
  (use "git add <file>..." to update what will be committed)
  (use "git restore <file>..." to discard changes in working directory)
	modified:   webots_ros2_driver/webots (new commits)
	modified:   webots_ros2_importer/webots_ros2_importer/urdf2webots (new commits)
```
Used this:
```
git submodule update --recursive webots_ros2_driver/webots
```
And then working tree was clean.

Now the mavic moves and i do not get the import error.

But i recognize i can't install my ros package with `--symlink-install`, then i get this this error:
```
[driver-2] ModuleNotFoundError: No module named 'webots_spot'
[driver-2] terminate called after throwing an instance of 'std::runtime_error'
[driver-2]   what():  The webots_spot.spot_driver.SpotDriver plugin cannot be found (C++ or Python).
[ERROR] [driver-2]: process has died [pid 78186, exit code -6, cmd '/home/max/webots_ws/install/webots_ros2_driver/lib/webots_ros2_driver/driver --ros-args --params-file /tmp/launch_params_bod_rxgw'].
```
But maybe i open another issue for that. Thanks again!!

closed 2022-01-25 by METEORITENMAX