cyberbotics/webots #4161: Segmentation

james-yoo, 2022-01-28:
**Describe the Bug**
I would like to get semantic segmentation result regarding of the 'RoadSegment'. However, I could not get result even if I placed the 'RoadSegment' to the Solid node with 'recognitionColors' as shown in picture below(You can see 'PedestrianCrossing' segment is displaying segmentation color yellow properly). 

**Steps to Reproduce**
1. Intall Webots R2022a
2. Open sample world project 'city.wbt'
3. Edit controller(autonomous_vehicle.c) to enable camera recognition and segmentation
4. Start simulation 

**Expected behavior**
RoadSegment should be displayed with recognitionColor.

**Screenshots**
[image]

**System**
 - Operating System: Linux Ubuntu 20.04
 - Graphics Card: NVIDIA GeForce RTX 2080 SUPER Max-Q

ad-daniel, 2022-01-28:
You don't need to put the `RoadSegment` inside a `Solid`, by default the Road proto (from which the `RoadSegment` is built from) defines recognition colors for the road lines, sidewalks and the asphalt itself. It just so happens that the color picked for the asphalt is also asphalt-colored. If you change this [line](https://github.com/cyberbotics/webots/blob/358d358e83e4d660987307632188832b5c61655a/projects/objects/road/protos/Road.proto#L256) to say green, it's much more noticeable

[image]

james-yoo, 2022-01-29:
Got it. Thanks!

closed 2022-01-29 by james-yoo