cyberbotics/webots #4168: Finish text does not render correctly on Wizard's button

BarceloChristian, 2022-01-28:
**Describe the Bug**
When running the `New project directory` wizard, in the last step, the button that says "Finish" does not render the text correctly when it's not selected.

**Steps to Reproduce**
1. Go to the Wizards menu
2. New project directory...
3. Follow the tutorial until reaching the end
4. Use TAB to move the selection out of the `Finish` button.
5. The text won't render correctly

**Expected behavior**
Buttons should render their text independently if they're selected or not

**Screenshots**
[image]

**System**
 - Operating System: Ubuntu 20.04
 - Graphics Card: NVIDIA GeForce 1650, 32 GB, Intel i5-10300H

**Additional context**
Add any other context about the problem here.

ad-daniel, 2022-01-28:
Personally I can't reproduce it on ubuntu 20.

If I drag the edges of the window to resize it then yes, while I'm dragging it does look like this:
[image]
but the moment I let go, the button returns to normal. This behavior only occurs for the dark themes, the classic one doesn't.

BenjaminDeleze, 2022-06-13:
I can reproduce it

PR #4629 (2022-06-13): Fix finish button
**Description**
Fix #4168 

@omichel it seems that you introduced those lines in https://github.com/cyberbotics/webots/pull/4020 to fix something on Windows. I didn't notice any changes by deleting these lines, but you know maybe where to look at.