cyberbotics/webots #4169: Weird boundingobject(collision) on the ElevationGrid

james-yoo, 2022-01-29:
**Describe the Bug**
BoundingObject is not aligned correctly when ElevationGrid is imported by 'osm_importer'.

**Steps to Reproduce**
1. Install Webots R2022a
2. Open world 'CH_Vens.wbt'
3. Add any object to the scene
4. Start simulation

**Expected behavior**
BoundingObject of the ElevationGrid should correctly matched to the Floor node(=appearance mesh).

**Screenshots**
BoundingObject of the ElevationGrid is not aligned as follow animation(you can see the contact point in the air):
[image]

**System**
 - Operating System:Windows 10,
 - Graphics Card: NVIDIA GeForce Titan 1080

ad-daniel, 2022-01-31:
Part of the issue is related to this https://github.com/cyberbotics/webots/issues/3884, meaning the location where the `boundingObject` is rendered (drawn) for elevation grid doesn't coincide where it is physically. But from your animation it seems the location is also wrong, or something else is causing it

james-yoo, 2022-02-02:
I would like to know how to fix this issue. Is it a problem of osm_importer?

ShuffleWire, 2023-09-06:
I just took a look at this bug, and it's still present in 2023b.

Crafted a more minimal example : 
[CH_Vens_simplified.wbt.zip](https://github.com/cyberbotics/webots/files/12537205/CH_Vens_simplified.wbt.zip)

But I suspect something is wrong with those huge Grid...

closed 2023-10-11 by omichel