cyberbotics/webots #4239: [SUMO Interface] SUMO does not recognize vehicle added in webots

tciuan, 2022-02-15:
**Describe the Bug**
I have connected SUMO and Webots using the SUMO interface. The SUMO generates serverals vehicles and it shows well in the webots. However, I add a vehicle in webots and I want the SUMO should also be able to detect the vehicle in generate the traffic flow.

**Expected behavior**
The SUMO should detect the vehicle added in webots and generate the trips which can avoid the vehicle. 

**Screenshots**
The red vehicle being crashed is added in webots directly. All the other vehicle are generated by SUMO. As shown in below figure, the SUMO does not recognize the red vehicle and crashed into the red one directly.

[image]

**System**
 - Operating System: [e.g., Windows 10]
 - Graphics Card: [e.g., NVIDIA GeForce  3070]

**Additional context**
Add any other context about the problem here.

ad-daniel, 2022-02-16:
I believe you need to give a DEF name to the red vehicle in the form: `WEBOTS_VEHICLE0` for sumo to be aware of it and avoid it (or at least try to, depending on where you spawn the other vehicles it might be too late to avoid a crash)

ad-daniel, 2022-02-24:
I will close this for now, feel free to re-open this if needed.

closed 2022-02-24 by ad-daniel

Yashraj19032001, 2022-04-21:
Hi, I am facing this issue in SUMO[image] ,[image] can u please guide me .
Traceback (most recent call last):
File "av_challenge_controller.py", line 3, in
from vehicle import Driver
File "D:\Pillai college\4th Year\Final Year Project\Webots\lib\controller\python37\vehicle.py", line 31, in
import _vehicle
ImportError: DLL load failed: The specified module could not be found.
WARNING: 'av_challenge_controller' controller exited with status: 1.
this is showing in my console.

Yashraj19032001, 2022-04-21:
Hi, I am facing this issue in SUMO INTERFACE, can u please guide me .
Traceback (most recent call last):
File "av_challenge_controller.py", line 3, in
from vehicle import Driver
File "D:\Pillai college\4th Year\Final Year Project\Webots\lib\controller\python37\vehicle.py", line 31, in
import _vehicle
ImportError: DLL load failed: The specified module could not be found.
WARNING: 'av_challenge_controller' controller exited with status: 1.
this is showing in my console.
[image]

omichel, 2022-04-21:
Can you open and run the sumo_interface_example.wbt file provided with Webots?
It seems your problem is not related with SUMO, but with Python.
May I ask you to try with Python 3.8 or 3.9 instead of 3.7?

omichel, 2022-04-21:
Can you try with Webots R2022a instead?

Yashraj19032001, 2022-04-21:
> Can you try with Webots R2022a instead?

It worked in python 3.8
I have one more doubt 
https://github.com/athishr88/Autonomous_vehicle
in this project 
in prallel parking and other scenario I am getting this error 
[image]

omichel, 2022-04-21:
You should probably pip install opencv for python 3.8.

Yashraj19032001, 2022-04-21:
It is showing like this 
[image]

Yashraj19032001, 2022-04-21:
[image]
I m facing this prolem 
[image]

omichel, 2022-04-26:
Did you managed to solve the problem?