cyberbotics/webots #4251: emcc compilation error

ad-daniel, 2022-02-18:
**Describe the Bug**
Fairly regularly for a week or two now I've been getting this error while compiling (master and develop branch alike). Cleansing has no effect, nor `git clean -dfx`. The error blocks the compilation, but if the same make command is executed after the error, the second time it compiles without complaining. Happens in the main PC, on the laptop, and in virtualbox.

```
# linking libwren.a
Error: Embedding "../../resources/wren/shaders/color_noise.frag" which is below the current directory "/home/daniel/webots_develop/src/wren". This is invalid since the current directory becomes the root that the generated code will see
emcc: error: '/home/daniel/webots_master/dependencies/emsdk/upstream/emscripten/tools/file_packager /home/daniel/webots_develop/resources/web/wwi/wrenjs.data --from-emcc --export-name=Module --preload ../../resources/wren/shaders' failed (returned 1)
make[1]: *** [/home/daniel/webots_develop/resources/Makefile.include:494: build/release/libwren.a] Error 1
make: *** [Makefile:112: webots_target] Error 2
```

BenjaminDeleze, 2022-02-18:
I am not able to reproduce it.

BenjaminDeleze, 2022-02-23:
I am now able to reproduce it systematically if I make `git clean -xdf`, `make`.

But only in one of my two webots directories

I will investigate

BenjaminDeleze, 2022-02-23:
I found why it works the second time but not the first one.
The workflow goes as follow:

- Make
- Check if wren need to be made for Webots
  - If yes, make wren AND THEN wrenjs
  - otherwise skip compilation of wren.

The first time we `make`, it build wren and then try to build wrenjs. But fail when trying to preload the shaders in wrenjs. The second time we `make`, wren is already built so we just skip its compilation (and thus we also skip wrenjs)

BenjaminDeleze, 2022-02-23:
The issue appears if using emscripten version 3.1.2 or later

PR #4276 (2022-02-28): Fix Wrenjs compilation
**Description**
Fix #4251
closed 2022-02-28 by BenjaminDeleze