cyberbotics/webots #4252: Error in the external controller of multi-process startup with Webots

svenkatachalam, 2022-02-18:
Hello All,

I am using webots to simulate robot operations for process sequencing. I have two robots in the environment and I tried writing external python program that controls the robot extern controllers. I made the steps as described in https://cyberbotics.com/doc/guide/running-extern-robot-controllers, but I am getting the error "Only one instance of the Robot class should be created". 

`if __name__ == "__main__":
    objUR10eLinearAxisControl = ur10e_linear_axis_control()  
    objUr10eControl = ur10e_control()
       
    os.environ["WEBOTS_ROBOT_NAME"]="UR10e"  
    p1=Process(target=objUR10eLinearAxisControl.move_robot, args=("0.157:0.157:1.57:1.57:1.57:0.157",))
    p1.start()
    p1.join()
    
    os.environ["WEBOTS_ROBOT_NAME"]="Linear_Axis_UR10e"
    p2=Process(target=objUR10eLinearAxisControl.move_ur10e_to_loc, args=("1",))
    p2.start()
    p2.join()`
       
To handle this error, I tried with creating an object in the "main" python program and pass this as parameter to the class being instantiated from the main program. I am not sure, if this is allowed, I tried. I am getting the error to access individual motors or sensors. I am using webots in Windows 10 computer with Python3.9. Thanks.

`if __name__ == "__main__":
    robot = Robot()
    objUR10eLinearAxisControl = ur10e_linear_axis_control(robot)  
    objUr10eControl = ur10e_control(robot)
       
    os.environ["WEBOTS_ROBOT_NAME"]="UR10e"  
    p1=Process(target=objUR10eLinearAxisControl.move_robot, args=("0.157:0.157:1.57:1.57:1.57:0.157",))
    p1.start()
    p1.join()
    
    os.environ["WEBOTS_ROBOT_NAME"]="Linear_Axis_UR10e"
    p2=Process(target=objUR10eLinearAxisControl.move_ur10e_to_loc, args=("1",))
    p2.start()
    p2.join()
[image]
[image]
`

ad-daniel, 2022-02-25:
I can't really reproduce the issue you're having, personally on linux it works fine.

1. Add
```
a = ur10e_linear_axis_control()
a.move_ur10e_to_loc(1)
```
and 
```
b = ur10e_control()
b.move_robot('0.1:0.2:0.3:0.2:0.1:0.2')
```
to `ur10e_linear_axis_controller.py` and `ur10e_controller.py` respectively so that they do something once launched.

2. make sure `WEBOTS_HOME` is defined and add this path to PYTHONPATH: `${WEBOTS_HOME}/lib/controller/python39`. Since you're on windows you'll need to define the `Path` variable as well: https://cyberbotics.com/doc/guide/running-extern-robot-controllers?tab-os=windows&tab-language=python
4. `export WEBOTS_ROBOT_NAME="Linear_Axis_UR10e"` and `export WEBOTS_ROBOT_NAME="UR10e"` respectively on each terminal which will run the controllers. The name here must be exactly what is written the in "name" field of each robot
5. Open simulation and run
6. Launch the two controllers

https://user-images.githubusercontent.com/44834743/155679397-1c3f7710-3dcf-4906-9bda-6587a19491a2.mp4

svenkatachalam, 2022-02-25:
Ok, Thank you for your time for investigating the issue. I think, it could be a problem due to OS (I am running on Windows). I have followed the steps from 2 and 3. I could run single controller from extern python program and I dont have any problem. Only with multiple robots, I am getting an error that "only one instance of the Robot class should be created". I tried with threading/multi processing module, and I got the error about "Swigpy objects cannot be pickled". I will try the program in Ubuntu computer and if I have any problem, I will create a new issue. For now, I will close this ticket. Thanks once again for your help.

closed 2022-02-25 by svenkatachalam