cyberbotics/webots #4280: Automatic exit of webots on ubuntu

HimGautam, 2022-02-24:
Hi, 
I was trying to use webots simulator to train and simulate quadruped robot by EPFL. While training, I activated optional render of polygon support. In the next run when I tried to accelerate the simulation, webots simulator crashed automatically with the following line on the console.
```/usr/local/bin/webots: line 90: 5289 Segmentation fault (core dumped) "$webots_home/bin/webots-bin" "$@"```

Upon close examination of the recent changes, I disable the rendering of support polygon and now webots runs smoothly.  I can share my work if needed.

ad-daniel, 2022-02-24:
If you can provide a minimal example that shows the crash it would definitely help. I've tried a few worlds and I'm unable to reproduce the issue. Something related to your specific robot might be the cause

HimGautam, 2022-02-25:
Hi,
It can also be local to my system. I am currently using Lenovo Legion Y540 with 8GB RAM and Nvidia GeForce GTX 1650 with 4GB. However, I am attaching the link my work [here](https://drive.google.com/file/d/1lCHzL_yBa65GEsLXH59HXMK7o0oB5t5C/view?usp=sharingl).

ygoumaz, 2022-03-04:
Hi,

I was able to reproduce your issue using your project in both develop and master branches. I identified the code line which throws the segmentation fault. It is located in `src/nodes/WbSolid.cpp` in `prePhysicsStep(double ms)`:
```cpp
for (i = 0; i < mJointChildren.size(); ++i)
```

The mJointChildren seems to be NULL at some point and calling the size() methods causes the crash. I suspect that the QVector object is getting deleted somewhere but I could not find where exactly.

Other observations: It seems to occur 99% of the time when the support polygon is containing 10 or more vertices. Also, I could only reproduce the issue when running in fast mode.

stefaniapedrazzi, 2022-06-01:
@[HimGautam](https://github.com/HimGautam) this issue should now be fixed in #4569 and the patch should be available in the nightly build of R2022a-rev1 starting from tomorrow and as well in the next release of Webots.
Please let us now if you should experience it again with the newest version of Webots.

PR #4569 (2022-06-01): Fix support polygon access out of bounds
Fix #4280.
If `numberOfContactPoints` value greater than `mSupportPolygon` size, `mSupportPolygon` was often accessed outside his bounds.
If `numberOfContactPoints` is lower than 32 (default size of the QVarLengthArray `mSupportPolygon`) nothing bad occurs, otherwise we may overwrite other WbSolid structures like the `mJointChildren` vector.
closed 2022-06-01 by stefaniapedrazzi